me-no-dev/ESPAsyncWebServer #820: webSocket.binaryAll sending ESP32-Cam data Buffer/Memory issues?

cgmckeever, 2020-08-06:
Im pretty sure I am just slamming out the buffer/memory, but I am not sure how to throttle it.  I've gone deep through issues and google but I havent come up with the solution. Is thee a way to check that a `binaryAll` has completed and do any clean up before sending the next? I tried adding a timer, but I still get crashes as 500ms - and even when it slightly works, the delay on the feed is bad.

```
 abort() was called at PC 0x4016063b on core 0
```
```
camera_fb_t *fb = NULL;

...

void loop() {
   fb = esp_camera_fb_get();
   webSocket.binaryAll(fb->buf, fb->len);
   esp_camera_fb_return(fb);
}
```

cgmckeever, 2020-08-06:
[I found part of a solution](https://github.com/me-no-dev/ESPAsyncWebServer/pull/821), it makes it tolerable, but not perfect

cgmckeever, 2020-08-07:
related 

```
PC: 0x400d2069: loop() at /Users/cgmckeever/Documents/Arduino/libraries/ESPAsyncWebServer/src/StringArray.h line 88
EXCVADDR: 0xbaad567c

Decoding stack results
0x400d2069: loop() at /Users/cgmckeever/Documents/Arduino/libraries/ESPAsyncWebServer/src/StringArray.h line 88
0x400e7b75: loopTask(void*) at /Users/cgmckeever/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 19
0x4008dd75: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

cgmckeever, 2020-08-07:
Also this [PR](https://github.com/me-no-dev/ESPAsyncWebServer/pull/822) helps with resolution

closed 2020-10-21 by stale[bot]

un0038998, 2021-09-12:
I faced the similar issues with AsyncWebSocket when sending huge data continuously .
I was sending binary image from esp32 cam board and each image was 5Kb .
So when we keep sending messages , the queue gets full if connection is slow or receiver is slow.
Then It consumes whole heap memory available for esp32 board (max heap available around 180kb) .
So I changed the WS_MAX_QUEUED_MESSAGES  parameter to 15 from 32. So the max messages queued will have the total size of 15*5kb = 75kb which is much less than total heap .
It resolved the issue for me and it never crashed again .
You can change this setting in AsyncWebSocket.H file in the library folder.

Check the size of the messages you are sending and set this parameter so that it does not exceed total heap memory.
Hope this might help.