me-no-dev/ESPAsyncWebServer #823: request->send(200, "/xyz","OK" );  intercept

ats3788, 2020-08-11:
Hi One last question is it possible to intercept    request->send(200, "/buttons","OK" );
on the Client Side with java script 
```
server.on("/color2", HTTP_GET, [] (AsyncWebServerRequest *request) {
    String inputMessage1;
    if (request->hasParam("rgba"))  {
      inputMessage1 = request->getParam("rgba")->value();
      }    else
    inputMessage1 = "No message sent - rgba";
      
   Serial.println(inputMessage1);
  

    request->send(200, "/buttons","OK" );
  });
```

lorol, 2020-08-13:
```
...
            var xhttp = new XMLHttpRequest();
            xhttp.onreadystatechange = function() {
                if (this.readyState == 4 && this.status == 200) {
                    document.getElementById(YOUR_ELEMENT_ID).innerHTML = this.responseText;
                }
            };
            xhttp.open("GET", /buttons, true);
            xhttp.send()
...
```

ats3788, 2020-08-17:
O THX I have to try this later . Wow what a wonderful  Library .

closed 2020-10-31 by stale[bot]