me-no-dev/ESPAsyncWebServer #824: Limited events

palmacas, 2020-08-11:
Hi,

I'm using `AsyncWebServer` and `AsyncEventSource` to show 12 different variables that are read from a couple environmental sensors. Each variable is updated every 20 seconds using `events.send`. However, only 8 of those variables are being updated automatically. The other variables doesn't change even if the measured values are different.

Is there any limit on the number of events that the handler can support?

ats3788, 2020-08-13:
I think you should post at first your Code

palmacas, 2020-08-23:
I have the following script in my html code:

```
<script>
    if (!!window.EventSource) {
    var source = new EventSource('/events');
    source.addEventListener('open', function(e) {console.log("Events Connected"); }, false);
    source.addEventListener('error', function(e) {if (e.target.readyState != EventSource.OPEN) {console.log("Events Disconnected"); } }, false);
    source.addEventListener('message', function(e) {console.log("message", e.data); }, false);
    source.addEventListener('temperature', function(e) {console.log("temperature", e.data); document.getElementById("temp").innerHTML = e.data; }, false);
    source.addEventListener('weather', function(e) {console.log("weather", e.data); document.getElementById("weather").innerHTML = e.data; }, false);
    source.addEventListener('feels_like', function(e) {console.log("feels_like", e.data); document.getElementById("feels_like").innerHTML = e.data; }, false);
    source.addEventListener('pressure', function(e) {console.log("pressure", e.data); document.getElementById("press").innerHTML = e.data; }, false);
    source.addEventListener('humidity', function(e) {console.log("humidity", e.data); document.getElementById("hum").innerHTML = e.data; }, false);
    source.addEventListener('wind', function(e) {console.log("wind", e.data); document.getElementById("wind").innerHTML = e.data; }, false);
    }
</script>
```
I created an event handler `server.addHandler(&events)` and I'm updating the events using this code on the `void loop()`:

```
getWeather();
events.send("ping", NULL, millis());
events.send(String(temperature).c_str(), "temperature", millis());
events.send(String(feels_like).c_str(), "feels_like", millis());
events.send(String(weather).c_str(), "weather", millis());
events.send(String(pressure).c_str(), "pressure", millis());
events.send(String(humidity).c_str(), "humidity", millis());
events.send(String(wind).c_str(), "wind", millis());
```

Looks like the limit of events per update is 8, as a workaround I created another event handler `server.addHandler(&local_events)` and I'm updating the other variables there.

closed 2020-11-06 by stale[bot]