me-no-dev/ESPAsyncWebServer #826: Gzip

arrow2020, 2020-08-18:
Hello, I have a question regarding the use of gzip in asyncwebserver, I am using the following code to open the index.html.gz but the browser does not automatically show me the index.html. What changes would I have to make to open the index.html automatically? I have another question, how to use gzip for my project if in it I am using css, main.js and thus compress all these files?

void InitServer()
{   
   server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html.gz").setCacheControl("max-age=300");
  server.onNotFound([](AsyncWebServerRequest *request) {
      request->send(400, "text/plain", "Not found");
   });
   server.begin();
   Serial.println("HTTP server started");
}

Pablo2048, 2020-08-19:
Set the default file for just "index.html". The server will search for compressed one automatically and add the needed header about gzipped file also...

MA-SHaikh-Project, 2020-09-30:
Change

closed 2020-12-13 by stale[bot]