me-no-dev/ESPAsyncWebServer #827: Server does not catch first HTTP GET request

andmarti1424, 2020-08-18:
Sometimes (not every time... Maybe after some period of inactivity) the server cant seem to handle the first HTTP GET request. But only the first one!
The browser stays idle waiting for a response.. until of course it times out.
If you send another HTTP GET just one second after the first one, the server answers without problem...

My code is a simple..

`  // Route / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
      Serial.printf("NEW GET\n");
      request->send(200, "text/html",
                       "<title>ESP2 wifi</title>"
                       "<body><h1>"
                       "Hello from ESP2 !!"
                       "</h1></body>");

  });
  // Start server
  server.begin();`

Thanks!

andmarti1424, 2020-08-18:
BTW, I experienced this behaviour in three different boards.

andmarti1424, 2020-09-04:
Yep. I tried it and some behaviour. Could you analyze this issue? Thanks again.

lorol, 2020-09-06:
@andmarti1424  Clear the cache of your browser, see why  ( https://www.f5.com/services/resources/white-papers/caching-behavior-of-web-browsers )  
You can try also your sketch w/o forcing the static IP. You can find the device IP by adding Serial.println(WiFi.localIP()); and looking at serial monitor.

andmarti1424, 2020-09-07:
@lorol, I understand but this does not seems the case, since I can reproduce this behaviour just using telnet.. Its something in the way the connection is handled after the socket is opened.. it never gets to the server.on() handler..

lorol, 2020-09-07:
Hi @andmarti1424 
I tested your sketch and everything works fine at my side. 
Absolutely every visit goes to  server.on() handler. I tried with Mozilla on WIn 10 64, as well as Safari i-phone, and page is served. In 3 min, in 20min and in 2h later
Differences from your sketch:
- I added    Serial.begin(115200); and Serial.setDebugOutput(true); at the beginning of setup()
- I added Serial.println("entered"); at server.on
- I used my fork of the library https://github.com/lorol/ESPAsyncWebServer
(It has many manually merged changes and additions, so it is not easy to track back and I will not going to verify details if it works for you)  :)

Hope that helps.

andmarti1424, 2020-09-10:
I am testing with your branch and will report back

andmarti1424, 2020-09-10:
No. I am afraid it also fails with your fork. Didnt add the Serial lines you added/modded.
Dont think that would make a difference.

lorol, 2020-09-10:
For me, this is your setup, router network. On my side it gets absolutely all requests processed.
Watching the serial and adding these lines will help - you may see things there.

closed 2020-11-23 by stale[bot]