me-no-dev/ESPAsyncWebServer #835: javascript client

mer30boy, 2020-08-25:
Hi 
I connected to this server socket via a JavaScript client by following code : 
`socket = new WebSocket('ws://192.168.1.2/ws');`

i want to do a simple authentication using  username and password in connecting.
It seems that JavaScript cannot do authentication in this way.
so I tried to use the [socket.io](https://www.npmjs.com/package/socket.io-client) in JavaScript. but encountered a CORS error while connecting.

How can I authentication using a JavaScript client and ESPAsyncWebServer socket server

lorol, 2020-08-25:
CORS:  https://github.com/me-no-dev/ESPAsyncWebServer/#adding-default-headers

WEBSOCKET authentication is tricky
https://www.freecodecamp.org/news/how-to-secure-your-websocket-connections-d0be0996c556/
https://devcenter.heroku.com/articles/websocket-security
https://stackoverflow.com/questions/22383089/is-it-possible-to-use-bearer-authentication-for-websocket-upgrade-requests

You can see in people' forks of this repository for authentication-related mods ... hope something may work for you:
https://github.com/sascha432/ESPAsyncWebServer
Also mine, a cookie-based on an early PR:  https://github.com/me-no-dev/ESPAsyncWebServer/pull/684
https://github.com/lorol/ESPAsyncWebServer

closed 2020-11-07 by stale[bot]

lorol, 2020-11-26:
@mer30boy 
You need following:
1) A login post request to create cookie, separate or within your main page:
https://github.com/lorol/ESPAsyncWebServer/blob/master/examples/SmartSwitch/data/login/index.htm#L15
2) Set cookie by server side:
https://github.com/lorol/ESPAsyncWebServer/blob/master/examples/SmartSwitch/SmartSwitch.ino#L596
3) Then html and ws both will use the same authentication through cookie already set
https://github.com/lorol/ESPAsyncWebServer/blob/master/examples/SmartSwitch/data/index.htm#L421
4) Modified library to be able to create a "myHandshake" filter
https://github.com/lorol/ESPAsyncWebServer/blob/master/examples/SmartSwitch/SmartSwitch.ino#L341
You can run the full example and modify it.
Even you want "only" ws, you need to handshake by setting auth pieces before connecting to it.