me-no-dev/ESPAsyncWebServer #838: HTTPS Support. Please!

johnkelton, 2020-08-29:
This library is awesome but we would love support for HTTPS. All modern browsers require HTTPS to access features like geolocation, PWA etc. Can someone point to a library based on this library or the developer's plan to include the support?

lorol, 2020-08-30:
:)  if this was a restaurant to order HTPPS and have it served to you it would be really awesome. We will order it as well.

johnkelton, 2020-08-30:
!?

lorol, 2020-08-30:
It is always good to search before asking. I just didn't need SSL yet, but on quick Google search found this:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/753

Please try and report, it may be valuable for the same community you mentioned.

BlueAndi, 2020-08-30:
https://github.com/me-no-dev/AsyncTCP/pull/90

johnkelton, 2020-08-31:
@lorol If one day you end up needing help with SSL, you will find yourself part of the same “community” I mentioned! We all start somewhere. 

When I have something Working on HTTPS, I will share with the community.

lorol, 2020-08-31:
@johnkelton  👍 +1 I wish you good luck and I would love to see your code!