me-no-dev/ESPAsyncWebServer #841: `request->send(... stuff...)` gets stuck

qcfe, 2020-09-07:
I'm writing an appliance that includes Wi-Fi network scanning on ESP32.
The Wi-Fi interface starts in `WIFI_MODE_APSTA`. On boot I try to connect to the last saved network, if present.
If within 15 secs there's no connectivity I invoke the `WiFi.disconnect()` method.
Within a POST call, I scan Wi-Fi networks and send them in JSON format.
The following code works perfectly, however in the following context it fails:
- A previously saved Wi-Fi network isn't reachable (switched off/out of range/etc)
- 15 seconds are elapsed from the startup 
Within this scenario, the ESP gets stuck on the last line and the wdt reboots the ESP32. If I disable the wdt with `esp_task_wdt_delete(NULL);` then it just stays stuck there until manual reset. However this does not happen the first time I call that API method. Most of the times it happens on the second or third time.

By digging into the code, it seems that inside `size_t AsyncClient::write(const char* data, size_t size, uint8_t apiflags)` the add() or send() method are the ones in which it hangs, especially around the LwIP calls, however I couldn't find anything else. 

If I removed the `WiFi.scanNetworks();` invocation it seems to work, but I can't see how this can influence that...

```
typedef StaticJsonDocument<2048> StaticJson;

 server.on("/api/wifiScan", HTTP_POST,
    [](AsyncWebServerRequest *request) {}, nullptr,
    [](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) {
      StaticJson res;

      bool isOnAP = WiFi.localIP() != request->client()->localIP();

      if (isOnAP) WiFi.disconnect();

      int n = WiFi.scanNetworks();
      JsonArray aps = res.createNestedArray("aps");

      for (int i = 0; i < n; ++i) {
        JsonObject ap = aps.createNestedObject();
        ap["ssid"] = WiFi.SSID(i);
        ap["rssi"] = WiFi.RSSI(i);
        ap["crypto"] = WiFi.encryptionType(i);
      }

      if (!WiFi.isConnected()) {
        bool hasWiFiSaved = trySavedWifiConnect();
        if (!isOnAP) return;

        if (hasWiFiSaved) {
          for (int i = 0; i < 7 && !WiFi.isConnected(); i++) {
            esp_task_wdt_reset(); // prevent wdt to kill the process
            delay(1000);
          }
        }
      }

      if (WiFi.isConnected()) {
        res["currentSSID"] = WiFi.SSID();
        res["cliIP"] = WiFi.localIP().toString();
        res["connected"] = true;
      } else res["connected"] = false;

      res["onAP"] = isOnAP;
      res["success"] = true;

      String ret;
      serializeJson(res, ret);
      request->send(200, "application/json", ret);
  });

bool trySavedWifiConnect() {
  WiFi.begin("unreachableSSID", "asdasd");
  return true;
}
```

lorol, 2020-09-07:
You cannot have delay() there, see https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember

qcfe, 2020-09-07:
Hmm I missed that point, thank you so much for pointing that out. 
However I guess this is because possible context switches between the two cores of the ESP32?
I mean, if I use a busy waiting loop it shouldn't be an issue I guess, given there's no such kind of advice in the readme...

closed 2020-11-21 by stale[bot]