me-no-dev/ESPAsyncWebServer #845: Send large webpage over 30kb

puby, 2020-09-13:
Hello everyone,

Does anyone know how to send out more than 30K webpage data ?

I have tried the following three methods, but unfortunately they all fail

_AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", index_html);
request->send_P(200, "text/html", index_html, processor);
AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", index_html, processor);_

Even if the dog is added, it is invalid
//---------------------------------------------------------------------------------------
   server.on("/tst", HTTP_GET, [](AsyncWebServerRequest * request) {
    
    AsyncResponseStream *response = request->beginResponseStream("text/html");
    response->print("Start ********************** ");

    for(int lop=0; lop<=1700; lop++){      
      response->print("<p>");
      response->println(lop);
      response->print("/<p>");
      esp_task_wdt_reset();
      }
    request->send(response);
//---------------------------------------------------------------------------------------

I tried the tracking program, "request->send(response);" ,This command will always induce WDT errors !

E (48565) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (48565) task_wdt:  - async_tcp (CPU 0/1)
E (48565) task_wdt: Tasks currently running:
E (48565) task_wdt: CPU 0: IDLE0
E (48565) task_wdt: CPU 1: IDLE1
E (48565) task_wdt: Aborting.
abort() was called at PC 0x400e431b on core 0

I also use Exception Decoder to check, The result seems to be caused by WTD.
//------------------------------------------------------------------------------------------------------
Decoding stack results
0x4008ce38: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008d069: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400e431b: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x40151ccf: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400e56cb: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008ad41: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x4008954d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
//------------------------------------------------------------------------------------------------------

On the LIB of WizNet W5100, It can use "server.println(F("<P>"));" to send the data one by one. this will have a very good advantage when sending large record data.
Instead of sending out the data and piece together at once, this will easily cause timeout or memory corruption.
Does ESPAsyncWebServer have a similar approach ? Or other alternatives ?

Thank.

lorol, 2020-09-13:
@puby  Yeah, it is is bit heavy this way ... The size is not a problem, rather the timing to process & flush. Try by changing 32 to 256 here: https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.cpp#L98
I am curious if it will help :)

puby, 2020-09-14:
@lorol 

Thank you for your assistance, I changed from 32 to 256 and it didn’t improve.
It caused another kind of error :

```
Decoding stack results
0x4008ce38: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008d069: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40144483: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401444ca: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x40143ddf: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x40143ece: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x40143e85: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400e1e97: cbuf::resize(unsigned int) at C:\Users\pharaoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp line 49
0x400e1edd: cbuf::resizeAdd(unsigned int) at C:\Users\pharaoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp line 35
0x400daf8c: AsyncResponseStream::write(unsigned char const*, unsigned int) at H:\PH_DOCX\Desktop\Arduino Files\arduino-1.8.13\libraries\ESPAsyncWebServer-master\src\WebResponses.cpp line 695
0x400e0f22: Print::write(char const*) at C:\Users\pharaoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h line 66
0x400e1005: Print::print(char const*) at C:\Users\pharaoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 89
0x400d14f5: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ) at H:\Cloud\Dropbox\Arduino\ESP32\EK_63/Server_on.ino line 19
0x400dc56d: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\pharaoh\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400da55d: AsyncWebServerRequest::_parseLine() at H:\PH_DOCX\Desktop\Arduino Files\arduino-1.8.13\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 588
0x400da629: AsyncWebServerRequest::_onData(void*, unsigned int) at H:\PH_DOCX\Desktop\Arduino Files\arduino-1.8.13\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 124
0x400da895: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ,  ,  ) at H:\PH_DOCX\Desktop\Arduino Files\arduino-1.8.13\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 77
0x400d7a8d: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at c:\users\pharaoh\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d7b35: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at H:\PH_DOCX\Desktop\Arduino Files\arduino-1.8.13\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 1210
0x400d8146: _async_service_task(void*) at H:\PH_DOCX\Desktop\Arduino Files\arduino-1.8.13\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 162
0x4008954d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

Where else can I make adjustments ?

lorol, 2020-09-14:
Sorry, I cannot help you. I can recommend you to use a buffer or a file (ram or flash) to fill your data and serve it at once instead of loop within the handler.

BlueAndi, 2020-09-14:
The code fragment in the inital post shows that the response is complete handeled inside the web request handler, which of course triggers the task wd after staying there too long.
Couldn't you only begin the response stream handling in the web request handler and continoue in the main loop?

puby, 2020-09-15:
@BlueAndi 

Thank you for participating in the discussion.
My dilemma is that I don’t know where to insert the dog feed command.

request->send(response);

I have traced it and it always crashes on this command, and the bottom layer of this command, I don't know where I can start to improve.

puby, 2020-09-15:
@lorol 

Even if it’s in the FLASH or SD card, It is unavoidable to use LOOP to combine and send out.
So I think this may be unsolvable.

Anyway, thank you very much.

lorol, 2020-09-15:
The solution would be to process your loop to feed data to a flash file or memory buffer outside the handler routine and  then serve when ready. 
Alternatively, you can use websockets

BlueAndi, 2020-09-15:
Thats why my recommendation was to do it in the main loop and not inside the web request handler. If you do such long taking things in a thread with dedicated priority (which you cant control), the probabilty is high to run in problems. Feeding the wd inside is just a workaround, as you may block other important threads with lower or equal priority.

puby, 2020-09-19:
@BlueAndi & @lorol 

Thank you very much for your suggestion, about "beginChunkedResponse"  it's extremely effective and very scary !

Even 200KB web pages can be easily transferred, and no accident.

``
const char html[] PROGMEM = R"rawliteral(<html><head><title> ......)rawliteral"; // <---- Content 200 KB

request->beginChunkedResponse("text/html",
                                       [](uint8_t* buffer, size_t strLen, size_t index)

while (len < (maxLen) && indexvReal < strLen(html)
{
len += sprintf(((char *)buffer + len),
}
 ......

Although this is perfect, but this is the performance under static char array.

What I want to know is, if my data is from a file "Log,txt" stored on the SD card, and I don’t want to read it and place it in the array first.
Because it will consume extra memory or cause memory crash. Instead, read it directly and send it out.

Could you please teach me and show me a code, how to implement this paragraph ?

lorol, 2020-09-19:
What do you want to do exactly? Is it a web browser at your client side? One or multiple clients?
Normally, a web page is asked by client and served at once by server. It needs some amount of memory anyway. 
You can fill the content to a file and leave servestatic to do the rest. But I believe in your case a websocket will work much better. You can keep appending data through it.

puby, 2020-09-19:
@lorol 

In fact, I am trying to complete a project, which can beautifully present the recorded marine meteorological data on the WebPage in the form of charts.

You can visit this website :

https://dygraphs.com/gallery/#g/resize

Just the JavaScript charting library it needs. It has 156KB, the record data to be sent is about 50KB, so it seems a bit challenging.

I have not found an example of websocket , so no way to start. Do you have relevant information to provide me ?
Grateful.

SimonMerrett, 2020-09-19:
@puby there are many examples of websockets with ESPAsyncWebServer, several by the Santos brothers. Here's one which I found really helpful. https://techtutorialsx.com/2018/08/14/esp32-async-http-web-server-websockets-introduction/

puby, 2020-09-19:
@SimonMerrett 

Thank you very much for your kind assistance, I will read carefully.

closed 2020-12-03 by stale[bot]