me-no-dev/ESPAsyncWebServer #846: Bind with Ethernet Port Only

maganuk, 2020-09-15:
In my project I have both a Wifi Connection and an Ethernet Connection. Is it possible to bind the webserver only to the ethernet interface? Currently I can make requests on both the WiFi and the Ethernet Interface.

zekageri, 2020-11-18:
I belive both is using the same interface. I'm not sure how can you use both at the same time.

ArminPP, 2020-12-16:
Hi there,
I wrote some code using this great library with a m5Stack. It uses 3 html files, bootstrap.js / .css and jquery and so on and it works like a charm.

[image]
Now I bought the Ethernet Unit (W5500 poe) and wanted to expand the code.
Unfortunately, I had to find out the hard way that ESPAsyncWebServer does not work with the Ethernet.h library.

Will it be possible to use the Ethernet.h library in the near future together with ESPAsyncWebServer ?

Many thanks
best regards, Armin

p.s.:
> I belive both is using the same interface. I'm not sure how can you use both at the same time.

Of course it is possible, I have a olimex Gateway Shield [https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware](url) and use WiFi and Ethernet at the same time (without ESPAsyncWebServer).

alainkovacs, 2020-12-29:
@maganuk could you please share with us how you were able to use this library with the ethernet interface? I have been trying to do exactly that for that past two days but I am unable to. Right now I don't care about doing it over the WiFi as that works, but I just want to achieve the same over Ethernet but it seems impossible. 
Thanks for the help beforehand and Happy New Year to everybody!

CurtRod, 2021-01-19:
The linked ESP32-GATEWAY (Olimex) uses a LAN8710 chip and not a W5500. That seems to work well with the espasycwebserver.

closed 2021-06-02 by stale[bot]