me-no-dev/ESPAsyncWebServer #847: restart after request-> send

unbreakmat, 2020-09-15:
Hi 

im progging an example like this:
- a request is sended 
- the server reboots

But the problem is when i send the request directly after that my esp reboots (this is the purpose) and my esp crashes with a gigantic error code. 
How can i make sure that the requestis sended and if this is OK then the esp reboots?

i'm fairly new to the library and read the faq but i hav eno clue?
Should i use a type of handler or a bool 'isSending' that block the reboot when sending (because set on true)  until it is set back to false (the sending is done) ?
could someonoe helpe me with this code ?

code:
```
request->send(200, FPSTR(HTTP_HEAD_CT), page);
rebootESP();

```
im stuck because it works asynchronious but i have the "block" the code in some way that is doesnt go to fast (which is now the problem)

Pablo2048, 2020-09-16:
Hi,
I'm using Ticker exactly for this case. You can try this:
```cpp
static Ticker deferred;
...
request->send(....);
deferred.once_ms(500, []() {
  ESP.restart();
});
```

unbreakmat, 2020-09-16:
Pablo2048 you would be a live savier if this works! i will test it out and let you know!!

lorol, 2020-09-16:
No need to time it...
See this: https://github.com/lorol/ESPAsyncWebServer/blob/737a3567b760e8a38600b1e4ac4f5045ed37b5ae/examples/SmartSwitch/SmartSwitch.ino#L653

unbreakmat, 2020-09-16:
Pablo how can i send a var in the defered once function? Now it says 'isOK not captured'
`
deferred.once_ms(500, []() {                        
    DEBUG_M("ESP", "Ticker passed");
    if (isOk) {
      stopAp();                                         
      connectToWiFi(false);                           
    }
  });
`

Pablo2048, 2020-09-17:
What are you trying to achieve? Where `isOk` came from? It wasn't mentioned in your first post..

unbreakmat, 2020-09-19:
It wasnt mentioned indeed, i added it while coding, it is declared above the ticker to do some checking, can i pass it inside the ticket?
Otherwise i have to write a ticker for true and for false, buth i want to write them in once

The only difference between them is that when isAp is true another function should also be activated in the ticker, so nOw i have 2 tickers but want to make them as one

Pablo2048, 2020-09-20:
Please post complete code, otherwise we are wasting time. Without seeing your code I can only think that your definition of `isOk` is not global or static so the variable doesn't exist when the lambda fires.

unbreakmat, 2020-09-20:
```
 bool isOk = handleParamSave(request);                         //check the post on error handling
  page += FPSTR(HTTP_FORM_RESPONS);                             //html form respons
  page.replace(FPSTR(T_v), isOk ? FPSTR(S_Success) : FPSTR(S_Unsuccess));

  page += isOk ? FPSTR(HTTP_FORM_SUCCESS) : FPSTR(HTTP_FORM_UNSUCCESS);

  page += compileFooter();
  request->send(200, FPSTR(HTTP_HEAD_CT), page);
  if (isOk) {
    if (get_mode() == WIFI_STA) {
      deferred.once_ms(500, []() {                              //defer 500ms
        DEBUG_M("ESP", "Ticker passed");
        connectToWiFi(false);                                   //reconnect in STA mode
      });
    } else if (get_mode() == WIFI_AP) {
      deferred.once_ms(500, []() {                              //defer 500ms
        DEBUG_M("ESP", "Ticker passed");
        stopAp();                                               //stop the AP - mode
        connectToWiFi(false);                                   //reconnect in STA mode
      });
    }
  }
```

So you see now i have 2 deferred onces. i want to make 1 like this:
deferred.once_ms(500, []() {                              //defer 500ms
        DEBUG_M("ESP", "Ticker passed");
        if(isAp){
           stopAp();                                               //stop the AP - mode
        }
        connectToWiFi(false);                                   //reconnect in STA mode
      });

Pablo2048, 2020-09-20:
So you have to made `isOk` available all the time. Try this:
```cpp
static bool isOk = handleParamSave(request);
```
then the deferred lambda function can access the isOk variable when needed.

unbreakmat, 2020-09-20:
Declaring it static worked, thanks pablo!

closed 2020-09-20 by unbreakmat