me-no-dev/ESPAsyncWebServer #850: Basic Authentication with Custom HTML Form and Ajax

waqast, 2020-09-21:
I want to login to server with jQuery Ajax call and custom HTML Form. Dont want to use the default Basic Authentication Form. I am able to authenticate at first call but on page refresh authentication does not persist and also websocket connection is failed.

**Arduino Code**
```
webSocket.setAuthentication("admin", "pass");

server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    Serial.println("page request");
    if ( request->authenticate("admin", "pass") ){
      Serial.println("authorized");
      request->send(SPIFFS, "/index.html", "text/html");
    } else {
      //return request->requestAuthentication(NULL, false);
      Serial.println("unothorized");
      request->send(SPIFFS, "/auth.html", "text/html");
    }
  });

  server.on("/signin", HTTP_GET, [](AsyncWebServerRequest *request){
    Serial.println("ajax request");
    if ( request->authenticate("admin", "pass") ){
      Serial.println("ajax authentication success");
      request->send(200, "application/json", "{\"message\": \"auth_success\"}");
    } else {
      Serial.println("ajax authentication fail");
      request->send(200, "application/json", "{\"message\": \"auth_fail\"}");
    }
  });
```

**JQuery Code**
```
$(function(){

    $('#btn_signin').click(function(e) {

        let username = $('#input_username').val();
        let password = $('#input_password').val();
        
        $.ajax({
            type: 'GET',
            url: '/signin',
            dataType: 'json',
            crossDomain: true,
            cache: false,
            xhrFields: {
                withCredentials: true
            },
            beforeSend: function(req) {
                req.setRequestHeader( 'Authorization', make_base_auth(username, password) );
            },
            success: function (res) {
                console.log(res);
            },
            error: function(XMLHttpRequest, textStatus, errorThrown) {
                alert(textStatus, errorThrown);
                console.log(errorThrown);
            },
        });

    });

});

function make_base_auth(username, password) {
    var token = username + ':' + password;
    var hash = btoa(token);
    return 'Basic ' + hash;
}
```
**Console Error**
```
WebSocket connection to 'ws://thermascan-5001.local/websocket' failed: HTTP Authentication failed; no valid credentials available
```

zekageri, 2020-11-23:
If you want it to "presist" you have to set a cookie on the ajax success callback about the user stuffs. Next request process the cookie on the header from the request and auto redirect to the index page if the cookie says that this user successfully logged in the last time. 

That's how i would do it

zekageri, 2020-11-23:
**Javascript BASIC example:**

```
function setCookie(cname, cvalue, exdays) {
        var d = new Date();
        d.setTime(d.getTime() + (exdays * 24 * 60 * 60 * 1000));
        var expires = "expires="+d.toUTCString();
        document.cookie = cname + "=" + cvalue + ";" + expires + ";path=/";
      }
      
   function getCookie(cname) {
        var name = cname + "=";
        var ca = document.cookie.split(';');
        for(var i = 0; i < ca.length; i++) {
          var c = ca[i];
          while (c.charAt(0) == ' ') {
            c = c.substring(1);
          }
          if (c.indexOf(name) == 0) {
            return c.substring(name.length, c.length);
          }
        }
        return "";
    }

$(function(){

    $('#btn_signin').click(function(e) {

        let username = $('#input_username').val();
        let password = $('#input_password').val();
        
        $.ajax({
            type: 'GET',
            url: '/signin',
            dataType: 'json',
            crossDomain: true,
            cache: false,
            xhrFields: {
                withCredentials: true
            },
            beforeSend: function(req) {
                req.setRequestHeader( 'Authorization', make_base_auth(username, password) );
            },
            success: function (res) {
                setCookie(res.UserName, true, 365);  // HERE YOU CAN SET A COOKIE FOR THE NEXT REQUEST
                console.log(res);
            },
            error: function(XMLHttpRequest, textStatus, errorThrown) {
                alert(textStatus, errorThrown);
                console.log(errorThrown);
            },
        });

    });

});
```

**ESP Server side:**

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    boolean Was_User_Here_Before = false;
    //get specific header by name (Compatibility)
    if(request->hasHeader("Cookie")){
      //Process User Cookies, if UserName is true than Was_User_Here_Before = true
    }else{
      Was_User_Here_Before = false;
    }

   if(Was_User_Here_Before){
    //Send Index.html
   }else{
     //Authenticate
   }
  });
```

closed 2021-02-07 by stale[bot]