me-no-dev/ESPAsyncWebServer #852: Captive portal on asyncWebServer

williamNizard, 2020-09-23:
Hi,
I'm really new in C++ and I want to understand the exemple for the captive portal,
I wan't to know how can I do to use that portal but without the link to the IP.

" void handleRequest(AsyncWebServerRequest *request) {
    AsyncResponseStream *response = request->beginResponseStream("text/html");
    response->print("<!DOCTYPE html><html><head><title>Captive Portal</title></head><body>");
    response->print("<p>This is out captive portal front page.</p>");
    response->printf("<p>You were trying to reach: http://%s%s</p>", request->host().c_str(), request->url().c_str());
    response->printf("<p>Try opening <a href='http://%s'>this link</a> instead</p>", WiFi.softAPIP().toString().c_str());
    response->print("</body></html>");
    request->send(response);
  }"
I really don't understand how can I go too my IP without the link in the printf.
thanks for the time you will spend to help me.

williamNizard, 2020-10-05:
Is there anyone here who can answer?

BlueAndi, 2020-10-05:
I don't understand the question, but if you connect to spawned access point and try to access any page, shouldn't you automatically be routed to the captive portal page?

williamNizard, 2020-10-07:
Yes it does but my question was: How can I do to remove the first page with the captive portal handler and replace it by my website?

williamNizard, 2020-10-07:
The function handleRequest from the exemple print a page with a link to the good IP. how can I do to remove that step and go to the good IP in the first place. thank you!

BlueAndi, 2020-10-07:
You can automatically redirect e.g. via: https://www.w3docs.com/snippets/html/how-to-redirect-a-web-page-in-html.html

williamNizard, 2020-10-08:
thanks you i'm going to try!!

closed 2020-12-22 by stale[bot]