me-no-dev/ESPAsyncWebServer #853: Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores

ivanrlg, 2020-09-26:
Hello, I am doing a project on a "nodemcu esp8266". I am using your library and SPIFFS where I have the html stored.

This project measures with an ultrasonic sensor and displays it in the browser.
in my Setup function I have this:

```
server.on (" / ", HTTP_GET, [] (AsyncWebServerRequest * request) {
        if (DEBUG_SW) Serial.println ("\ n /");
        / * if (! request-> authenticate (mConfig.http_username, mConfig.http_password))
                return request-> requestAuthentication (); * /
        request-> send (SPIFFS, "/index.html");
    });

    server.on ("/ level", HTTP_GET, [] (AsyncWebServerRequest * request) {
        isBusy = true;
        request-> send_P (200, "text / plain", ReadTank1Ptc (). c_str ());
        isBusy = false;
    });

    server.on ("/ CurrentVol", HTTP_GET, [] (AsyncWebServerRequest * request) {
        isBusy = true;
        request-> send_P (200, "text / plain", ReadTank1Vol (). c_str ());
        isBusy = false;
    });

```

```
String ReadTank1Ptc () {
    float h = Ptc1;
    if (isnan (h)) {
        Serial.println ("Failed to read from readTank1Ptc sensor!");
        return "0";
    }
    else
    {
        return floatAsString (h);
    }
}

String ReadTank1Vol () {
    float h = VolumeTank1;
    if (isnan (h))
    {
        Serial.println ("Failed to read from readTank1Vol sensor!");
        return "0";
    }
    else
    {
        return floatAsString (h);
    }
} 
```
On my page I have these functions that are executed every 15 and 20 seconds

```
setInterval(function ( ) {
        var xhttp = new XMLHttpRequest();
        xhttp.onreadystatechange = function() {
          if (this.readyState == 4 && this.status == 200) {
            document.getElementById("myOutput").value=parseFloat(this.responseText);
          }
        };
        xhttp.open("GET", "/CurrentVol", true);
        xhttp.send();
      }, 15000 ) ;
      
      setInterval(function ( ) {
        var xhttp = new XMLHttpRequest();
        xhttp.onreadystatechange = function() {
[... 53 lines cut ...]
0x40208e8c: AsyncWebServerRequest::beginResponse(int, String const&, String const&) at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 731
0x40214f48: String::copy(char const*, unsigned int) at C:\Users\ivanr\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 209
0x40208ea7: AsyncWebServerRequest::send(int, String const&, String const&) at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 772
0x40205c70: __lambda17::operator()(AsyncWebServerRequest*) at C:\Users\ivanr\Dropbox\PROYECTO PROGRAMACION\Arduino\SmartLevel/SmartLevel.ino line 467
0x40214d85: String::invalidate() at C:\Users\ivanr\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 140
0x40201340: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at c:\users\ivanr\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40205cac: std::_Function_handler ::_M_invoke(const std::_Any_data &, AsyncWebServerRequest *) at c:\users\ivanr\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4020ba61: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\ivanr\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2464
0x4020bed4: AsyncWebServer::_attachHandler(AsyncWebServerRequest*) at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebServer.cpp line 116
0x40209ba5: AsyncWebServerRequest::_parseLine() at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 581
0x40215f40: String::trim() at C:\Users\ivanr\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 828
0x40209df0: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 124
0x40100300: millis() at C:\Users\ivanr\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40209e49: std::_Function_handler ::_M_invoke(const std::_Any_data &, void *, AsyncClient *, void *, unsigned int) at c:\users\ivanr\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40210812: AsyncClient::_recv(std::shared_ptr &, tcp_pcb*, pbuf*, long) at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp line 649
0x402109ba: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at C:\Users\ivanr\OneDrive\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp line 738
0x4022b968: tcp_input at core/tcp_in.c line 501
0x40100aa4: malloc(size_t) at C:\Users\ivanr\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x4023142a: ip4_input at core/ipv4/ip4.c line 1467
0x40100a6f: free(void*) at C:\Users\ivanr\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x40228405: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40228224: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 469
0x4025c4da: ethernet_input at glue-esp/lwip-esp.c line 365
0x4025c4eb: ethernet_input at glue-esp/lwip-esp.c line 373
```

lorol, 2020-09-26:
Try instead of:
 request->send_P(200, "text / plain", ReadTank1Ptc (). c_str ());
use
 request-> send(200, "text / plain", ReadTank1Ptc ());

ivanrlg, 2020-09-27:
> Try instead of:
> request->send_P(200, "text / plain", ReadTank1Ptc (). c_str ());
> use
> request-> send(200, "text / plain", ReadTank1Ptc ());

Hello Thank you very much for your answer, unfortunately, it did not work to change the send_p to send, what else could I try? Greetings.

closed 2020-12-10 by stale[bot]