me-no-dev/ESPAsyncWebServer #855: Webserver crashes on connection with I2C

AdriVerhoef, 2020-09-28:
Hello folks,

I'm making an ESP32 balancing Robot with a web server in order to configure various parameters and will attempt plotting several parameters as well. I've created a custom PCB with an ESp32, 2 DRV8825 stepper motor drivers, an MPU6050 IMU, a I2C OLED display and an UART to USB converter. I've stopped using the I2C display as the refresh times are to long for the MPU6050 sample rate, since I use the blocking I2C Wire implementation. (Both the I2C OLED display and the MPU6050 are on two different I2C buses)

With the web server I'm trying to implement in the ESP32 I've come across the problem that with the MPU6050 on a sample rate of 200Hz, it crashes when I try to load the web server with my browser (Firefox). It does not always reset, but quite often. I use an external gpio interrupt to detect a new sample on the interrupt pin of the MPU and then read the data. I've tried using the millis function for timing or detaching the interrupt pin in the interrupt handler, but that doesn't help. What does help is if I remove the I2C call to read the MPU6050 registers (14 bytes ~0.5ms @ 400kHz), then the web server works like a charm.

With an oscilloscope I have checked that the MPU6050 read frequency and the I2C communication is correct.
In order to alleviate this problem, I've tried making a task for reading the MPU and tested with pinning the tasks to either core, but to no avail.

It seems to me that if the web server interrupts the I2C communication and the SPIFFS is read, the ESP32 resets. I hope someone has had a similar problem and knows how to solve this or send me in the right direction.

If any additional information or code is required, let met know.

```
void WEB::begin(const char * ssidKey, const char * passwordKey, AwsEventHandler handler)
{
  Preferences prefsSSID, preffsPASSWORD;
  if(ssidKey == NULL || passwordKey == NULL)
    return;

  server    = new AsyncWebServer(80);
  websocket = new AsyncWebSocket("/datastream");

  // Initialize the SPIFFS. Format it when the init fails
  Serial.println("SPIFF");
  if(!SPIFFS.begin()){
    Serial.println("An Error has occurred while mounting SPIFFS");
  }

  // Set your Static IP address, ensure that if the IP address is changed, the websocket IP address in the index page is changed as well.
  Serial.println("Static IP");
  local_IP = IPAddress(192, 168, 1, 184);
  gateway  = IPAddress(192, 168, 1, 1);
  subnet   = IPAddress(255, 255, 0, 0); 

  // Configures static IP address
  #if 1
    if (!WiFi.config(local_IP, gateway, subnet)){
      Serial.println("STA Failed to configure");
    }
  #endif

  Serial.println("Retrieve ssid & password");
  prefsSSID.begin(MAIN_KEY, false);
  prefsSSID.getString(ssidKey, ssid, STRING_LENGHT);
  prefsSSID.end();

  preffsPASSWORD.begin(MAIN_KEY, false);
  preffsPASSWORD.getString(passwordKey, passWord, STRING_LENGHT);
  preffsPASSWORD.end();

  Serial.printf("Wifi SSID:=%s, password=%s\n\r", ssid, passWord);
  WiFi.begin(ssid, passWord);

  const uint8_t connectResult = WiFi.waitForConnectResult();
  Serial.printf("WiFi result %u\n\r", connectResult);
  if(connectResult == WL_CONNECTED)
  {
    IP  = WiFi.localIP().toString();
    MAC = WiFi.macAddress();
    Serial.printf("Connected to wifi IP=%s, MAC=%s\n\r", IP.c_str(), MAC.c_str());
    wifiConnected = true;
  }
  else
  {
    Serial.printf("Connecting to WiFi failed\n\r");
    wifiConnected = false;
    return;
  }

  // Tell the ESP how to handle specific incoming HTTP requests
	server->on("/", HTTP_GET, [](AsyncWebServerRequest *request){
		request->send(SPIFFS, "/index.html", "text/html");
	});

  // Tell the WebSocket to execute the function onWsEvent() on each WebSocket event
	websocket->onEvent(handler);
	server->addHandler(websocket);

  server->begin();
}
```

ClaesClaes, 2020-10-11:
Hi,
I can report similar problem as yours (although not using SPIFF but FFat instead). In my case the I2C communication on my ESP32 was interrupted (see error dump below) when I uploaded a file to the ESPAsyncWebServer, which in turn led to a restart. I traced it to a function (drawStatus) that updated my I2C OLED when an upload was in progress. This function was called inside the ESPAsyncWebServer uploadHandler. 

I solved it by removing the call to "drawStatus" and inserted a public flag (state.downloading) in the uploadHandler and check the flag inside the main loop where I placed the call to "drawStatus". I consider this a workaround in my particular case so not really solving your problem which seems to be incompatibility with simultaneous ESPAsyncWebServer / I2C activity at least on ESP32. But it could be worth a while to check that you are not interacting with I2C inside any of your ESPAsyncWebServer handlers, which seems to need undivided attention.

My ESPAsyncWebServer upload handler: 
```
void onUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {   if (!index) {   
    state.downloading = true;
    Serial.printf("UploadStart: %s\n", filename.c_str());
    if (!filename.startsWith("/")) filename = "/" + filename;
    if (FFat.exists(filename)) FFat.remove(filename);
    uploadFile = FFat.open(filename, "w");
    filename = String();
  }
  for (size_t i = 0; i < len; i++) {
    uploadFile.write(data[i]);
  }
  if (final) {
    uploadFile.close();
    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index + len);
    state.downloading = false;
  }
}
```

My (pseudo) main loop:

```
void loop() {
  doStuff();
  if (state.downloading) {
        u8g2.clearBuffer();
        drawStatus(0, 43, DWNLD);
        u8g2.sendBuffer();
      }
}
```

Error dump:
```
10:31:53.112 -> [I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
10:31:53.183 -> [D][esp32-hal-i2c.c:1345] i2cProcQueue():  Gross Timeout Dead start=0xc7c3b, end=0xc7c6d, =50, max=50 error=1
10:31:53.183 -> [E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbed38
10:31:53.183 -> [I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
10:31:53.183 -> [I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb843c
10:31:53.183 -> [I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
10:31:53.217 -> [I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
10:31:53.217 -> [I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
10:31:53.217 -> [I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
10:31:53.217 -> [I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=200
10:31:53.217 -> [I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
10:31:53.217 -> [I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffdafb8
10:31:53.217 -> [I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
10:31:53.250 -> [I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
10:31:53.250 -> [I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-18
10:31:53.250 -> [I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
10:31:53.250 -> [I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
10:31:53.250 -> [I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
10:31:53.250 -> [I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
10:31:53.250 -> [I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
10:31:53.317 -> [D][esp32-hal-i2c.c:1336] i2cProcQueue():  Busy Timeout start=0xc7cc5, end=0xc7cf7, =50, max=50 error=1
10:31:53.317 -> [E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbed38
10:31:53.317 -> [I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
10:31:53.317 -> [I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb843c
10:31:53.355 -> [I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
10:31:53.355 -> [I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
10:31:53.355 -> [I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
10:31:53.355 -> [I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
10:31:53.355 -> [I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=200
[... 35 lines cut ...]
10:31:53.608 -> [I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
10:31:53.608 -> [I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=3
10:31:53.641 -> [I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=200
10:31:53.641 -> [I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
10:31:53.641 -> [I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffdafb8
10:31:53.641 -> [I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
10:31:53.641 -> [I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
10:31:53.641 -> [I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
10:31:53.641 -> [I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
10:31:53.675 -> [I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
10:31:53.675 -> [I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
10:31:53.675 -> [I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
10:31:53.675 -> Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
10:31:53.675 -> Core 1 register dump:
10:31:53.675 -> PC      : 0x400e4467  PS      : 0x00060530  A0      : 0x800e491a  A1      : 0x3ffb1d70  
10:31:53.675 -> A2      : 0x00000020  A3      : 0x00000000  A4      : 0x00000032  A5      : 0x3ffbed38  
10:31:53.709 -> A6      : 0x60013000  A7      : 0x0003f7a0  A8      : 0x800e445d  A9      : 0x3ffb1d50  
10:31:53.709 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffdac70  A13     : 0x3ffc34a8  
[... cut ...]
10:31:53.709 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
10:31:53.743 -> 
10:31:53.743 -> Backtrace: 0x400e4467:0x3ffb1d70 0x400e4917:0x3ffb1db0 0x400e25a9:0x3ffb1dd0 0x400e25dd:0x3ffb1df0 0x400e2639:0x3ffb1e10 0x400e14f6:0x3ffb1e30 0x4015b2a1:0x3ffb1e50 0x400e1e3d:0x3ffb1e70 0x400e1f0b:0x3ffb1e90 0x4015b2e6:0x3ffb1eb0 0x400e2031:0x3ffb1ed0 0x400e2051:0x3ffb1ef0 0x4015b341:0x3ffb1f10 0x400e15ad:0x3ffb1f40 0x400e15e1:0x3ffb1f60 0x400d74f3:0x3ffb1f80 0x400e76bd:0x3ffb1fb0 0x40089605:0x3ffb1fd0
10:31:53.780 -> 
10:31:53.780 -> Rebooting...
```

AdriVerhoef, 2020-11-19:
The problem was caused by calling another function in the gpio callback function, which was not in IRAM. Once I removed this function call, the crash would no longer occur. It would only crash when the gpio interrupt, web server and I2C where active. If I disabled one of these, the crash wouldn't occur any more. This is why I had previously incorrectly assumed that the I2C communication was the cause.

closed 2020-11-19 by AdriVerhoef