me-no-dev/ESPAsyncWebServer #863: Send large webpage from PROGMEM containing templates with '%' sign in it (CSS part of page)

OPatrick, 2020-10-17:
Hello,
Since the % sign is used as token for the templating, how can I make an HTML page which contain % signs not being parsed, like so: 
width: 100%;
Thanks

Pablo2048, 2020-10-17:
Already discussed here https://github.com/me-no-dev/ESPAsyncWebServer/issues/644

OPatrick, 2020-10-17:
Yes, I red it.
Tried putting #define TEMPLATE_PLACEHOLDER '$' as first line of main script (before including lib) . Do not work for me.
Replacing '%' signed by '%%' were applicable, works. But it's a weird way because HTML content needs to be modified.

Pablo2048, 2020-10-17:
Escaping character with the second one is standard way in many template engine IMO so nothing weird here. If `#define` way doesn't work for you you can modify the library source code, and put the new character here https://github.com/me-no-dev/ESPAsyncWebServer/blob/f13685ee97675be2ac9502d177d3024ebc49c1e0/src/WebResponseImpl.h#L63

patrickbussmann, 2020-11-23:
@Pablo2048 I defined it at the top of my script:

    #define TEMPLATE_PLACEHOLDER '$'

But its not working.

Im only using the `$` char for templating - but the `%` chars also for CSS styles.

Where should I define it so that it works? Thanks in Advance!

Pablo2048, 2020-11-24:
You have to modify `WebResponseImpl.h` directly. Just to put the new define somewhere inside script is not enough unfortunately.

closed 2021-02-07 by stale[bot]