me-no-dev/ESPAsyncWebServer #870: How to disconnect a client from websockets

rikki78, 2020-10-23:
Hi,
I have a project where the ESP updates its webpage dynamically by sending json strings to it over a websocket. Now sometimes the connection fails for some reason, and the ESP starts buffering the strings.
This works for some time but at a point it will give the error
```
ERROR: Too many messages queued<CR>
```
So I'm using a ping-pong mechanism to check which client is not responding so I can close the connection. 
I've tried 
```ws.cleanupClients();
  for (const auto& c: wsClients) {
    for (uint8_t i = 0; i < DEFAULT_MAX_WS_CLIENTS; i++)
    {
      if (c->id() == wsClientStatus[i].id)
      {
        if (wsClientStatus[i].status == WS_CL_PING)  // ping outstanding, no pong received
        {
          wsClientStatus[i].status = WS_CL_INACTIVE; 
          wsClientStatus[i].id = 0; 
          Debug_V_printf("WS Client %d inactive, closing\n", c->id());
          c->close();
        }
      }
    }
  }
```
but when I check the status a bit later it is still connected (but not responding).

Does anyone has an idea how to close this connection?

zekageri, 2020-11-04:
Almost exactly the same problem happening with me. The connection remains connected even after the client closing it. That way my esp restarts sometimes and i don't know what to do.

I have seen this workaround which iam using currently.

```
void CleanUP_WS(int Msec){
  if(millis() - Start_Time_Millis >= Msec){
   Start_Time_Millis = millis();
   ws.cleanupClients();
  }
}

  void loop(){
    CleanUP_WS(1000);
  }
```

BlueAndi, 2020-11-04:
May be related to #802 and #836.

zekageri, 2020-11-09:
I'm thinking of a solution where the esp constantly ping-pongs with the clients ( Javascript ) and if one is not responding in x msec than close that connection or notify the user about communication issues.

closed 2021-01-23 by stale[bot]