me-no-dev/ESPAsyncWebServer #872: On Posting Large Data onRequestBody called twice

biswaKL, 2020-10-28:
server.onRequestBody(handleBodyUpload);

I am trying to post a large json from POSTMAN. If the JSON is large. The **handleBodyUpload** is getting called twice.

Please help me in resolving it

Bomexxuk, 2020-10-29:
Hi, I had the same issue yesterday. 

Yes on large files, the request execute twice because **size_t len** is maximum 251 assuming you have JSON size of 310, then the request execute twice. First you will have to covert the **uint8_t data** into a String and determine if this is the first half of you object.  As you sending JSON, the first char in the sequence will be '{' . Create a global variable where you can store the first half of the JSON string. Once you done this you will need the 2nd half of the sequence. If the first char is not '{' , add this part of the JSON to the global variable.

Once its completed use your deserializeJson method to convert your String to JSON object. Or modify the library so it takes in bigger data.

BlueAndi, 2020-11-12:
https://github.com/me-no-dev/ESPAsyncWebServer#body-data-handling

closed 2021-01-14 by biswaKL