me-no-dev/ESPAsyncWebServer #878: How to print Content?

forna91, 2020-11-06:
Sorry guys, if I'm writing here I have finish the hopes to find solution.

I recive a POST from the webserver with some json, I don0t whant to handle the json but I want just print out the body of the request but I don't find any simple way to do it.

just request->ContentType and Lenght, nothing simple to handle the content as a String?

BlueAndi, 2020-11-06:
Use onRequestBody() and consider it can be called several times for the whole body. Push every fragment in a string and print it. But to avoid to reassmble it by yourself, use the provided JSON body handling and print the received JSON objects:

https://github.com/me-no-dev/ESPAsyncWebServer#json-body-handling-with-arduinojson

forna91, 2020-11-07:
It's uncomorensive for me..

```
  const size_t capacity = JSON_OBJECT_SIZE(10);
  String datasetup;
  DynamicJsonDocument json(capacity);
AsyncCallbackJsonWebHandler* handler = new AsyncCallbackJsonWebHandler("/rest/endpoint", [](AsyncWebServerRequest *request, JsonVariant &json) {
  JsonObject& jsonObj = json.as<JsonObject>();
  // ...
});
server.addHandler(handler);
```

I've tried this, but I get:

 error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson6152_0000010::ObjectRef&}' from an rvalue of type 'ArduinoJson6152_0000010::VariantAs<ArduinoJson6152_0000010::ObjectRef>::type {aka ArduinoJson6152_0000010::ObjectRef}'

   JsonObject& jsonObj = json.as<JsonObject>();

What I'm missing?

forna91, 2020-11-07:
ok find the way, just error when i change the addres

closed 2020-11-07 by forna91