me-no-dev/ESPAsyncWebServer #879: SPIFFSeditor file iterator not compatible with LittleFS

sticilface, 2020-11-07:
Probably goes without saying that the clue is in the name, but still... 

I needed to modify my file iterator to be able to call recursively for directory structures.  As you can see below, the files list contains a directory which the editor displays as a file. 
[image]
[image]

I have a working fix for esp8266 that looks like this

```cpp SPIFFSEditor.h
 void _fileListIterator(FS & fs, const char * dirName, std::function<void(File & f)> Cb ); 
```

```cpp SPIFFSEditor.cpp
void SPIFFSEditor::_fileListIterator(FS & fs, const char * dirName, std::function<void(File & f)> Cb )
{
    Dir dir = fs.openDir(dirName);
    while (dir.next()) {
      if (dir.isFile()) {
        File f =  dir.openFile("r");
        if (Cb && f) {
          Cb(f); 
          f.close(); 
        }
      } else {
        _fileListIterator(fs, dir.fileName().c_str() , Cb);
      }
    }
}
```

and finally.. the business end changed to... 

```cpp
void SPIFFSEditor::handleRequest(AsyncWebServerRequest *request){
  if(_username.length() && _password.length() && !request->authenticate(_username.c_str(), _password.c_str()))
    return request->requestAuthentication();

  if(request->method() == HTTP_GET){
    if(request->hasParam("list")){
      String path = request->getParam("list")->value();
      path = String();
      String output = "[";
      _fileListIterator(_fs, path.c_str(), [&output](File &f) { 
	        if (output != "[") output += ',';
	        output += "{\"type\":\"";
	        output += "file";
	        output += "\",\"name\":\"";
	        output += String(f.fullName());
	        output += "\",\"size\":";
	        output += String(f.size());
	        output += "}";
      });
      output += "]";
      request->send(200, "application/json", output);
      output = String();
    } ...
```

However, I've not used SPIFFS very much with ESP32 and have not developed on it in a very long time.  How different are the FS APIs and does ESP32 support LittleFS now / is planning to? 

I can make a pull request, should probably change it to FSEditor as well rather than SPIFFS.. 

Thoughts?

lorol, 2020-11-07:
@sticilface 
See a workaround also it in [my fork](https://github.com/lorol/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp), 
The editor is almost an application than library and works with specific html part. I am not sure of the best way to be made universal.

sticilface, 2020-11-07:
ah i see.  A bit more work is involved, i also forgot the exclude list!   I'm currently having a bit of a nightmare trying to bring all my ESPs up to date from late 2018 so i can't really work on it now.  Is there a plan to bring littleFS to ESP32?

Pablo2048, 2020-11-07:
https://github.com/lorol/LITTLEFS ;-)

lorol, 2020-11-07:
and ... it is considered for the next generation of the core (2.x):
https://github.com/espressif/arduino-esp32/tree/esp32s2

lorol, 2020-11-07:
Not easy to be made ideally compatible to SPIFFS, to esp8266 LittleFS implementation and the esp32 IDF / vfs concept

sticilface, 2020-11-07:
mmm.. that's what I was thinking should be the goal.  I found it very useful for development and tweaking things without having to upload new files constantly.  

The other way.. might be to just make the FSEditor totally independent of the filesystem and it takes a linking class and uses callbacks to populate the data... this can be customised to each use case.  Moving forward...

It would also allow it to be used with SD cards and the likes. 

```cpp
FSwebhandle = & webServer.addHandler(new FSEditor( SPIFFSEditorimpl, "user", "pass" ));
```

or 

```cpp
FSwebhandle = & webServer.addHandler(new FSEditor( LittleFSEditorimpl, "user", "pass" ));
```

sticilface, 2020-11-16:
> @sticilface
> See a workaround also it in [my fork](https://github.com/lorol/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp),
> The editor is almost an application than library and works with specific html part. (I have some particular additions, not good for a PR)
> I am not sure of the best way to be made more universal. You can elaborate and PR as a cleaner solution. Just keep in mind other than listing, the creation of non existing new file on esp32 LITTLEFS needs folders of the path to be pre-made manually by mkdir in contrast of esp8266 way, deletion is tricky - folders again rmdir ... and/or you need to determine what editor should do with (empty) folders. You can run [ESP_AsyncFSBrowser](https://github.com/lorol/ESPAsyncWebServer/tree/master/examples/ESP_AsyncFSBrowser) variant to see in practice in your browser /edit

My JS is rubbish.  I've had a look at your fork.  what are the changes in edit.htm that you have made?  

I'm currently refactoring using templates in a mostly transparent way. 

I'm also wondering if this just needs to be a operate lib.  or a submodule

sticilface, 2020-11-16:
@me-no-dev 

I'm writing a version of this that should be much more flexible, be platform and FS independent just requiring an interface class...  

I'm trying to allow multiple instances, so permit setting the URL of the editor in the constructor.  

unfortunately the `/edit` is hardcoded into the html 

here
https://github.com/me-no-dev/ESPAsyncWebServer/blob/f6fff3f91ebf45b91ca4cff2460d2febd61f1e27/src/edit.htm#L542
and 
https://github.com/me-no-dev/ESPAsyncWebServer/blob/f6fff3f91ebf45b91ca4cff2460d2febd61f1e27/src/edit.htm#L554

I really have no idea with JS unfortunately, and can't really test it either.

sticilface, 2020-11-16:
Ok scratch that previous comment.  I've managed to do it, i've now got an independent lib based on SPIFFSEditor that 

- has independent implementations for each device/file format.  My main thinking with this is to easily allow serving of an editor for files on SD cards, multiple partitions etc...  Currently it does depend on `fs::FS` and `fs::File` especially with the exclude list stuff but i 'should' be able to migrate it out and then it can work with anything.  I've not tackled ESP32 yet but should be easy.  There is quite a bit of code duplication which I can maybe get rid of by using some inheritance. 

- support for multiple instances of `FSEditor` so the following works in the same sketch
```
 server.addHandler(new FSEditor_ESP8266_SPIFFS    ("editSFS",  SPIFFS  ));
 server.addHandler(new FSEditor_ESP8266_LITTLEFS("/editLFS", LittleFS ));
```

I've actually written a lib called FSTools that is waiting for merge with the ESP8266 Core that allows some weird and wonderful FS stuff like converting between file system types and mounting two different partition types at the same time. 

https://github.com/esp8266/Arduino/pull/7696

- Modified the code to attempt to read an /edit.htm file from the file system then if it is not present then serve the static PROGMEM one.  

Currently there is not "much" difference functionally between the implementations, and as i mentioned the ESP32 is missing.  I could drop the dependence on `File` and `FS` which would allow some other weird uses such as virtual FS and maybe tag editing of the Preferences (nvs)??

I'd be interested on your thoughts?    

https://github.com/sticilface/FSEditor

sticilface, 2021-01-16:
Any thoughts?

closed 2021-01-31 by stale[bot]

embedded-creations, 2021-02-02:
This issue really should still be open.  @sticilface your solution seems on the right track but I haven't had time to really dig into it.  It would be nice if FSEditor had at least a basic README

embedded-creations, 2021-08-05:
I'm finally evaluating @lorol's fork for LittleFS compatibility (and planning to add support for SD and multiple FS sources simultaneously).  Some notes in case it helps anyone here:

I'm not sure but I think the SPIFFSEditor example might be broken on the fork. A working example that uses the updated editor is the SmartSwitch example. I got the editor in the example to work by:

flashing SmartSwitch example
ESP8266 LittleFS Data Upload
Find the network it serves, and add wifi credentials via the portal
Login to the root page, password "switch"
Now /edit works