me-no-dev/ESPAsyncWebServer #880: Multiple authentication

MassiPi, 2020-11-13:
Hello,
my esp is serving a webpage and a websocket connection. I'm looking for a way to manage more than one authentication credentials. 
For the webserver access it seems not to be a problem, i can manage more than one and i can manage it at request level (so i can check some requests and other no).
but what about the websocket? if i got it right, i can only "setAuthentication" on the websocket server when this is started, and with a single credentials pair!
Is there a way to manage authentication when the websocket connection is estabilished, like for the webserver?
thanks

lorol, 2020-11-13:
@MassiPi 
You can see my tweaks that may help you:
https://github.com/lorol/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.h#L324
https://github.com/lorol/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.cpp#L1277

Based on:
https://github.com/me-no-dev/ESPAsyncWebServer/pull/684

Then an example usage:
https://github.com/lorol/ESPAsyncWebServer/blob/master/examples/SmartSwitch/SmartSwitch.ino#L341
https://github.com/lorol/ESPAsyncWebServer/blob/master/examples/SmartSwitch/SmartSwitch.ino#L571

MassiPi, 2020-11-21:
hello @lorol and many thanks for the answer :)
sadly, i'm not sure i understood how to use it.
could you help me a little bit more?
i need to use the modified class, and this is ok, 
but can i use your "handshake" for multiple normal auth, and not cookie based?
thanks

lorol, 2020-11-22:
Hi @MassiPi 
In general websocket does not standardize basic auth, at least for old revisions / browsers. 
https://stackoverflow.com/a/7732613/2343527
I had problems with some devices and gave up on trying to use ws with base auth. Either I had to leave it open or it doesn't allow.
If you want to use just several different passwords and you want write a custom login page,  cans share some quick stuff to start with, but again in my case it uses Cookies. 
Another way you can see in the project here: https://github.com/rjwats/esp8266-react/tree/master/lib/framework
But this again doesn't work on old devices

MassiPi, 2020-11-24:
thank you again @lorol , very helpful and interesting.
Sadly, i'm not that agile in working with cookies when i have no PHP in my hands, but i'll try to understand something from your examples. I cant fully understand why basic auth should not work, where actually the websocket server can be secured with that approach (the setAuth is doing this, right?), but as far as i can understand the ws server is not exposing single requests, so i'll need another way.
Thanks again

MassiPi, 2020-11-25:
hello @lorol 
i took time to look at your modification and i think i partially understood it :)
as i said before, the point was the ability to access the single request time by time (like for the webserver), and you did it.
My question is why you simply did not add request to the event handler callback, wouldn't it be easier and cleaner?
btw, it seems with the handshake i can manage the auth by request, and this is good
thanks again, i'll also try to simply extend the event handler callback with the request..
regards

closed 2021-02-08 by stale[bot]