me-no-dev/ESPAsyncWebServer #882: WDT get reset, no delay, only a big loop to build the response

hitech95, 2020-11-14:
Hi, I have the folowing code that is triggering a wtchdog reset when there are a lot of networks available...
Am I doing something really stupid? I dont have any delay or stuff like that.

Is there something like yield() to reset the watchdog?
```
void ESPManager::handleWifiScan(AsyncWebServerRequest *request)
{
  bool refresh = request->hasArg(F("refresh"));
  bool scanning = false;

  StaticJsonDocument<1024> doc;
  JsonObject root = doc.to<JsonObject>();

  JsonArray networks = root.createNestedArray("availableNetworks");

  if (refresh || WiFi_scanNetworks() < 0)
  {
    WiFi_scanNetworks(refresh);
    scanning = true;
  }
  else
  {
    int n = WiFi_scanNetworks();
    if (n == 0)
    {
      DEBUG_WM(F("No networks found"));
    }
    else
    {
      DEBUG_WM(n, F("networks found"));
    }

    // build networks list
   // WDT get reset while here
    for (int i = 0; i < n; i++)
    {
      int rssiperc = getRSSIasQuality(WiFi.RSSI(i));
      uint8_t enc_type = WiFi.encryptionType(i);

      if (_minimumQuality == -1 || _minimumQuality < rssiperc)
      {
        JsonObject item = networks.createNestedObject();

        item["ssid"] = WiFi.SSID(i);
        item["encryption"] = encryptionTypeStr(enc_type);
        item["open"] = enc_type == WM_WIFIOPEN;
        item["quality"] = rssiperc;
        item["signal_strength"] = int(round(map(rssiperc, 0, 100, 1, 4)));
        item["rssi"] = WiFi.RSSI(i);
      }
    }
  }
  root["scannng"] = scanning;
  root["totalNetworks"] = (scanning) ? -1 : _numNetworks;

  AsyncResponseStream *response = request->beginResponseStream("application/json");
  serializeJson(doc, *response);
  request->send(response);
}

int8_t ESPManager::WiFi_scanNetworks()
{
  return WiFi_scanNetworks(false);
}

int8_t ESPManager::WiFi_scanNetworks(bool force)
{
  if (WiFi.scanComplete() == WIFI_SCAN_RUNNING)
  {
    return WIFI_SCAN_RUNNING;
  }

  if (force || _numNetworks == 0)
  {
    WiFi.scanDelete();
    _numNetworks = 0;

    using namespace std::placeholders; // for `_1`
    WiFi.scanNetworksAsync(std::bind(&ESPManager::WiFi_scanComplete, this, _1));
    return WiFi.scanComplete();
  }
  else
  {
    return WiFi.scanComplete();
  }
}

void ESPManager::WiFi_scanComplete(int networksFound)
{
  _numNetworks = networksFound;
}

int ESPManager::getRSSIasQuality(int RSSI)
{
  int quality = 0;

  if (RSSI <= -100)
  {
    quality = 0;
  }
  else if (RSSI >= -50)
  {
    quality = 100;
  }
  else
  {
    quality = 2 * (RSSI + 100);
  }
  return quality;
}

ul-gh, 2020-11-14:
Hi,

since the ESPManager::handleWifiScan(AsyncWebServerRequest *request) has the request object, this seems to run in the async_tcp task.

And indeed, if this is not disabled, the FreeRTOS task watchdog is enabled for this task, that is, while the events are handled (and disabled while waitng for network or ohter events):

From AsyncTCP.cpp:

```
static void _async_service_task(void *pvParameters){
    lwip_event_packet_t * packet = NULL;
    for (;;) {
        if(_get_async_event(&packet)){
#if CONFIG_ASYNC_TCP_USE_WDT
            if(esp_task_wdt_add(NULL) != ESP_OK){
                log_e("Failed to add async task to WDT");
            }
#endif
            _handle_async_event(packet);
#if CONFIG_ASYNC_TCP_USE_WDT
            if(esp_task_wdt_delete(NULL) != ESP_OK){
                log_e("Failed to remove loop task from WDT");
            }
#endif
        }

    } // for(;;)
    vTaskDelete(NULL);
    _async_service_task_handle = NULL;
}
```

So from your loop, you /could/ just call esp_task_wdt_reset() (#include esp_task_wdt.h) which should do the trick.

But I if you have such long delays in your handler, this will not only be a performance issue, but could have some unwanted side-effects, for example trigger the dead-lock issue #876 if there are other events piling up during this time.

So maybe it is better to have the really slow parts running in a separate task and not in the async web server callback, and communicate the results with a FreeRTOS queue.

Best, Ulrich

hitech95, 2020-11-15:
I'm new on this hw. I,m mainly a AVR guy.
I've forgot to say that I'm runnig on an esp8266, does it also have the RTOS in the background?

ul-gh, 2020-11-15:
Ooops. No. On the ESP8266, there is no RTOS.

I guess you could then just call the wdtFeed() from the loop and see if any problems arise.

Maybe still better, you could do the scan for each of the found networks individually in the main loop() and then poll for the finished results from the web page.

Or try to find out which part of the network scan takes too long and leave it out (getRSSIasQuality() ?)

hitech95, 2020-11-15:
> Maybe still better, you could do the scan for each of the found networks individually in the main loop() and then poll for the finished results from the web page.

Actually that method does not scan unless you specify the refresh argument, and the it basically return, the scan is run with asynk mode. Th results must be pulled by calling again the same route a few seconds later. (if you spam, it just doesn't do anything)

> Or try to find out which part of the network scan takes too long and leave it out (getRSSIasQuality() ?)
The method is asp privided, it is a basic match for converting a number to percentage. So I don't think it will take a lot.

As you see I'm not doing something compute heavy (excluding the json). For that reason I had no idea why WDT was triggered.

I'll try with the `wdtFeed()` call and hopefully it will get solved. (As reference the scan return more than 20 networks, when there are less the issue is quite hard to trigger)

BlueAndi, 2020-11-15:
Note, on the esp8266 there is not AsyncTcp running, but [ESPAsyncTcp](https://github.com/me-no-dev/ESPAsyncTCP).
How does the crash dump looks like? Note, the JSON document uses 1k from the stack, this may cause a problem.

ul-gh, 2020-11-15:
I need now to add, a lot of what is reported in issue #825 very much looks like the same issue.
See e.g. https://github.com/me-no-dev/ESPAsyncWebServer/issues/825#issuecomment-682110052

So fixing this will likely also fix #825.

hitech95, 2020-11-15:
If you have some other suggestion to test out I can do that.
@BlueAndi, I'll use heap if you think that ArduinoJSON could create issues.

closed 2021-01-30 by stale[bot]