me-no-dev/ESPAsyncWebServer #884: Concurrent access of AsyncEventSourceClient::_messageQueue causes memory access violation

ul-gh, 2020-11-16:
Hi,

in contrast to any normal AsyncWebHandler derivative which runs in the async_tcp task only:

When using an AsyncEventSource instance on ESP32 platform, the AsyncEventSourceClient::_messageQueue is supposed to be accessed by both the AsyncEventSource callbacks running in async_tcp task and any application task calling AsyncEventsource::send().

This causes dereferencing of invalid pointers in AsyncEventSourceClient::_runQueue() when the linked list is iterated over while being changed from another task. Although this might happen infrequently, this is a crash/DOS/safety issue.

This was already adressed by Arjan Filius by adding mutex locking of access to the AsyncEventSource _messageQueue (see iafilius/ESPAsyncWebServer/commit/d924de1 etc.)

With this report I am suggesting a pull request introducing the mutex lock wherever the linked list is iterated over, with slightly changed position as in ArFi commits.

Plus, the same locking mechanism is introduced where the AsyncEventSource::_clients list is iterated over as the public API is also supposed to be called from different tasks.

ul-gh, 2021-01-17:
Will there be any official fix for this?

brunojoyal, 2021-04-26:
> Will there be any official fix for this?

I would also like to know this... Thank you

closed 2021-07-11 by stale[bot]