me-no-dev/ESPAsyncWebServer #890: Camera Async Web Server Chunked Response

iot-objx, 2020-11-18:
Hello, 

I have been trying to port the [original](https://github.com/espressif/esp32-camera#jpeg-http-stream) example to an Async version using the ESPAsyncWebServer library. 

I am using the ESP Arduino core and my configuration is
- Board: ESP32-Cam
- Arduino 1.8.13

So far, I am stuck at translating 

```
if(res == ESP_OK){
    size_t hlen = snprintf((char *)part_buf, 64, _STREAM_PART, _jpg_buf_len);
    res = httpd_resp_send_chunk(req, (const char *)part_buf, hlen);
}
if(res == ESP_OK){
    res = httpd_resp_send_chunk(req, (const char *)_jpg_buf, _jpg_buf_len);
}
if(res == ESP_OK){
    res = httpd_resp_send_chunk(req, _STREAM_BOUNDARY, strlen(_STREAM_BOUNDARY));
}
```

And this is what I have tried so far. 

```
AsyncWebServerResponse *response = NULL;
size_t len = snprintf((char *)part_buf, 64, _STREAM_PART, _jpg_buf_len);
response = request->beginChunkedResponse(_STREAM_CONTENT_TYPE, [part_buf, len](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
    memcpy(buffer, part_buf, maxLen);
    if (index < len)
    {
        return (index + maxLen > len) ? (len - index) : maxLen;
    }
    return 0;
});

len = _jpg_buf_len;
request->sendChunked([_jpg_buf, len](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
    memcpy(buffer, _jpg_buf, maxLen);
    if (index < len)
    {
        return (index + maxLen > len) ? (len - index) : maxLen;
    }
    return 0;
});

len = strlen(_STREAM_BOUNDARY);
request->sendChunked([_STREAM_BOUNDARY, len](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
    memcpy(buffer, _STREAM_BOUNDARY, maxLen);
    if (index < len)
    {
        return (index + maxLen > len) ? (len - index) : maxLen;
     }
     return 0;
});
```
I have also tried playing with the content type on the above snippet. No luck so far. And, I get the following errors

- `Failed to load resource: net::ERR_INVALID_CHUNKED_ENCODING` on the browser side
- Occasional memory issues on the ESP32-Cam side: `Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.` and `Watchdog Timouts`.

It's important to note that these code snippets run in a `while` loop. 

I am waiting for directions and ideas. Thank you in advance!

me-no-dev, 2020-11-18:
have a look here: https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3

closed 2020-11-18 by me-no-dev

iot-objx, 2020-11-18:
Thank you for your super-fast response. This kind of solves what I am after. 

I was wondering how this continuous stream of jpeg performs compared to the original chunked implementation? I noticed very rare stuttering on the images although I don't mind that.