me-no-dev/ESPAsyncWebServer #893: ESP32 reboots with multi heap because of websockets

zekageri, 2020-11-24:
I'm sending a websocket message every 1 sec. And some random times the esp restarts with corrupt heap.
Here is the backtrace :

CORRUPT HEAP: multi_heap.c:187 detected at 0x3ffd7814
abort() was called at PC 0x4008d9f6 on core 1

`Backtrace: 0x4008e9b0:0x3ffd8720 0x4008ebe1:0x3ffd8740 0x4008d9f6:0x3ffd8760 0x4008dfa5:0x3ffd8780 0x4008e5be:0x3ffd87a0 0x400855da:0x3ffd87c0 0x40087039:0x3ffd87e0 0x4000bec7:0x3ffd8800 0x4015e4ad:0x3ffd8820 0x400e3de9:0x3ffd8840 0x400e4265:0x3ffd8860 0x400e4bd9:0x3ffd8890 0x400e4bf3:0x3ffd88b0 0x400e4c13:0x3ffd88d0 0x400d3332:0x3ffd88f0 0x400d62aa:0x3ffd8910 0x400d9f34:0x3ffd8a60 0x4008acd9:0x3ffd8a80`

```
  #0  0x4008e9b0:0x3ffd8720 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd8740 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4008d9f6:0x3ffd8760 in multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) get_prev_free_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:187
  #3  0x4008dfa5:0x3ffd8780 in multi_heap_free_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
  #4  0x4008e5be:0x3ffd87a0 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #5  0x400855da:0x3ffd87c0 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #6  0x40087039:0x3ffd87e0 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #7  0x4000bec7:0x3ffd8800 in ?? ??:0
  #8  0x4015e4ad:0x3ffd8820 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #9  0x400e3de9:0x3ffd8840 in std::_Function_handler<void (AsyncWebSocketMessageBuffer* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketMessageBuffer*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessageBuffer* const&) at lib/ESPAsyncWebServer/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871        
  #10 0x400e4265:0x3ffd8860 in std::function<void (AsyncWebSocketMessageBuffer* const&)>::operator()(AsyncWebSocketMessageBuffer* const&) const at lib/ESPAsyncWebServer/src/StringArray.h:53
      (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::remove(AsyncWebSocketMessageBuffer* const&) at lib/ESPAsyncWebServer/src/StringArray.h:128
      (inlined by) AsyncWebSocket::_cleanBuffers() at lib/ESPAsyncWebServer/src/AsyncWebSocket.cpp:1237
  #11 0x400e4bd9:0x3ffd8890 in AsyncWebSocket::textAll(AsyncWebSocketMessageBuffer*) at lib/ESPAsyncWebServer/src/StringArray.h:53  #12 0x400e4bf3:0x3ffd88b0 in AsyncWebSocket::textAll(char const*, unsigned int) at lib/ESPAsyncWebServer/src/StringArray.h:53
  #13 0x400e4c13:0x3ffd88d0 in AsyncWebSocket::textAll(String const&) at lib/ESPAsyncWebServer/src/StringArray.h:53
  #14 0x400d3332:0x3ffd88f0 in _ZL10Send_Async6StringS_$constprop$471 at lib/modbus_esp8266/src/Modbus.h:237
  #15 0x400d62aa:0x3ffd8910 in ESP_Time::Assemble_Time() at lib/modbus_esp8266/src/Modbus.h:237
  #16 0x400d9f34:0x3ffd8a60 in ESP_Time::Tick() at lib/modbus_esp8266/src/Modbus.h:237
      (inlined by) Time_System(void*) at src/Own_Headers/ESP_Time.h:147
  #17 0x4008acd9:0x3ffd8a80 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

[Here is the file at PASTEBIN.COM where the websocket collapses at random](https://pastebin.com/RTuXMaYm)

jharames, 2020-11-29:
Im seeing something similar, in my usage.  Im at the mercy of great libraries, as Im not a programmer.  I dont really know how to debug or fix the issue noted, but would love to help--- if I new how.
ecoding stack results
0x4008c4bc: commonErrorHandler_dump at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 509
0x4008c6ed: panicHandler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 259
0x401387bb: _GLOBAL__sub_I__ZNSt12ctype_bynameIcEC2ERKSsj() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/locale_facets.h line 1879
0x40138802: _GLOBAL__sub_I__ZNSt12ctype_bynameIcEC2ERKSsj() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/locale_facets_nonio.h line 318
0x40138475: std::get_terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 92
0x401381f8: std::__throw_logic_error(char const*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/functexcept.cc line 74
0x400d79cd: AsyncWebServer::on(char const*, unsigned char, std::function ) at c:\users\luv2s\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 1974
0x400d1726: loop() at C:\Users\luv2s\OneDrive\Documents\Arduino\Glue_tft_0\GLUE_TFT_v1.0/GLUE_TFT_v1.0.ino line 249
0x400dc62d: uart_on_apb_change at C:\Users\luv2s\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-uart.c line 388
0x40088c05: pxPortInitialiseStack at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 218

zekageri, 2020-12-01:
Your issue is not because of the websockets but asyncwebserver. My problem is still open. :/

zekageri, 2020-12-04:
```
CORRUPT HEAP: Bad head at 0x3ffd789c. Expected 0xabba1234 got 0x00000000
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4012cc73 on core 0

Backtrace: 0x4008e9b0:0x3ffd64c0 0x4008ebe1:0x3ffd64e0 0x4012cc73:0x3ffd6500 0x4008e5b5:0x3ffd6530 0x400855da:0x3ffd6550 0x40087039:0x3ffd6570 0x4000bec7:0x3ffd6590 0x4015f21d:0x3ffd65b0 0x400e4ae5:0x3ffd65d0 0x400e4f61:0x3ffd65f0 0x400e5f1d:0x3ffd6620 0x400e5f5d:0x3ffd6650 0x400e2bd5:0x3ffd6670 0x400e2d80:0x3ffd66a0 0x400e34b1:0x3ffd66c0 0x4008acd9:0x3ffd66f0
  #0  0x4008e9b0:0x3ffd64c0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd64e0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707      
  #2  0x4012cc73:0x3ffd6500 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x4008e5b5:0x3ffd6530 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #4  0x400855da:0x3ffd6550 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #5  0x40087039:0x3ffd6570 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42 
  #6  0x4000bec7:0x3ffd6590 in ?? ??:0
  #7  0x4015f21d:0x3ffd65b0 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #8  0x400e4ae5:0x3ffd65d0 in std::_Function_handler<void (AsyncWebSocketMessageBuffer* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketMessageBuffer*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessageBuffer* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #9  0x400e4f61:0x3ffd65f0 in std::function<void (AsyncWebSocketMessageBuffer* const&)>::operator()(AsyncWebSocketMessageBuffer* const&) const at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::remove(AsyncWebSocketMessageBuffer* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:128
      (inlined by) AsyncWebSocket::_cleanBuffers() at lib/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:1237
  #10 0x400e5f1d:0x3ffd6620 in AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #11 0x400e5f5d:0x3ffd6650 in std::_Function_handler<void (void*, AsyncClient*, unsigned int, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, unsigned int, unsigned int)#4}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, unsigned int&&, AsyncClient*&&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #12 0x400e2bd5:0x3ffd6670 in std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>::operator()(void*, AsyncClient*, unsigned int, unsigned int) const at lib\AsyncTCP\src/AsyncTCP.cpp:1116
      (inlined by) AsyncClient::_sent(tcp_pcb*, unsigned short) at lib\AsyncTCP\src/AsyncTCP.cpp:917
  #13 0x400e2d80:0x3ffd66a0 in AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
  #14 0x400e34b1:0x3ffd66c0 in _async_service_task(void*) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
      (inlined by) _async_service_task at lib\AsyncTCP\src/AsyncTCP.cpp:197 
  #15 0x4008acd9:0x3ffd66f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

zekageri, 2020-12-04:
It seems to me so far that if i don't send Strings but chars it is working properly. I'm running it 20 mins straight with 7 clients on the websocket and no problem so far.

zekageri, 2020-12-04:
OK 8 client and crash:

```
assertion "pbuf_free: p->ref > 0" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x4012cccb on core 1

Backtrace: 0x4008e9b0:0x3ffd65d0 0x4008ebe1:0x3ffd65f0 0x4012cccb:0x3ffd6610 0x40147c2b:0x3ffd6640 0x400e2cdf:0x3ffd6660 0x400e2d49:0x3ffd66a0 0x400e34d5:0x3ffd66c0 0x4008acd9:0x3ffd66f0
  #0  0x4008e9b0:0x3ffd65d0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd65f0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707      
  #2  0x4012cccb:0x3ffd6610 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x40147c2b:0x3ffd6640 in pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:1331
  #4  0x400e2cdf:0x3ffd6660 in AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
  #5  0x400e2d49:0x3ffd66a0 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, 
signed char) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
  #6  0x400e34d5:0x3ffd66c0 in _async_service_task(void*) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
      (inlined by) _async_service_task at lib\AsyncTCP\src/AsyncTCP.cpp:197 
  #7  0x4008acd9:0x3ffd66f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

zekageri, 2020-12-04:
Corrupt heap again at random. This is different crash report than the String one. I use chars to send on the websockets and that problem fixed i think.
Now it is corrupt heap again at random times...

```
CORRUPT HEAP: Bad tail at 0x3ffd7d9c. Expected 0xbaad5678 got 0xbaad5676
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4012cccb on core 1

Backtrace: 0x4008e9b0:0x3ffd6470 0x4008ebe1:0x3ffd6490 0x4012cccb:0x3ffd64b0 0x4008e5b5:0x3ffd64e0 0x400855da:0x3ffd6500 0x40087039:0x3ffd6520 0x4000bec7:0x3ffd6540 0x4015f275:0x3ffd6560 0x400e45e1:0x3ffd6580 0x4019427b:0x3ffd65a0 0x400e56f3:0x3ffd65c0 0x400e573d:0x3ffd65e0 0x400e5ddc:0x3ffd6600 0x400e5e05:0x3ffd6620 0x400e3032:0x3ffd6640 0x400e33b6:0x3ffd6670 0x400e33c9:0x3ffd66a0 0x400e350b:0x3ffd66c0 0x4008acd9:0x3ffd66f0
  #0  0x4008e9b0:0x3ffd6470 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd6490 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707      
  #2  0x4012cccb:0x3ffd64b0 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x4008e5b5:0x3ffd64e0 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #4  0x400855da:0x3ffd6500 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #5  0x40087039:0x3ffd6520 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42 
  #6  0x4000bec7:0x3ffd6540 in ?? ??:0
  #7  0x4015f275:0x3ffd6560 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #8  0x400e45e1:0x3ffd6580 in AsyncWebSocketMultiMessage::~AsyncWebSocketMultiMessage() at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #9  0x4019427b:0x3ffd65a0 in std::_Function_handler<void (AsyncWebSocketMessage* const&), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(AsyncWebSocketMessage*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessage* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #10 0x400e56f3:0x3ffd65c0 in std::function<void (AsyncWebSocketMessage* const&)>::operator()(AsyncWebSocketMessage* const&) const at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #11 0x400e573d:0x3ffd65e0 in LinkedList<AsyncWebSocketMessage*, LinkedListNode>::remove(AsyncWebSocketMessage* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) AsyncWebSocketClient::_runQueue() at lib/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:539
  #12 0x400e5ddc:0x3ffd6600 in AsyncWebSocketClient::_onPoll() at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #13 0x400e5e05:0x3ffd6620 in std::_Function_handler<void (void*, AsyncClient*), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #14 0x400e3032:0x3ffd6640 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at lib\AsyncTCP\src/AsyncTCP.cpp:1116   
  #15 0x400e33b6:0x3ffd6670 in AsyncClient::_poll(tcp_pcb*) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
  #16 0x400e33c9:0x3ffd66a0 in AsyncClient::_s_poll(void*, tcp_pcb*) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
  #17 0x400e350b:0x3ffd66c0 in _async_service_task(void*) at lib\AsyncTCP\src/AsyncTCP.cpp:1116
      (inlined by) _async_service_task at lib\AsyncTCP\src/AsyncTCP.cpp:197 
  #18 0x4008acd9:0x3ffd66f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

serkanp, 2020-12-07:
i had something like that.. 
i found that, if multiuser sends data at the same time, esp32 crashes.. 
so i managed it with QueueArray.. 
when any message received from any client, i store their messages in a QueueArray

i used something like this (i dont remember how i implemented but idea is someting like this):
struct queuehelper
```
{
    QueueArray<char> queue;
    unsigned long lastReceivedTime;
};

//lets limit users, but you may use LinkedList... (in struct you may add client.id() also)
queuehelper queues[6];
```

then on each message, i enquee messages of each client.. 
on a seperate loop, i check lastReceivedTime and if no message received in 250 ms from that client,i dequee message and process , and do my logic.. so i manage each message buffer one by one.. 
for me it worked without crash..

so dont forget, esp32 is a very limited device.. you must be very carefull on c++ , any mistake on char, String, char length etc.. 
it crashes.. 

or...
 you may put your code in try.. catch... block and see what happens.. it might not crash :)

zekageri, 2020-12-08:
The async server already uses queues for the messages. In fact the esp does not get any socket messages, just the connection request. The esp only sends messages to clients but it does that every second. Sockets are essential for me so if the client for some reason gets a disconnect from the socket it try to connect back again after 3 sec. I have problems with more than 3 clients. One or two is pretty stable. 

So should I queue the esp's socket messages before send to the async queue?

zekageri, 2020-12-08:
```
***********Connected Client**************
Count of Clients: 6
Client IP: 92.118.176.228
Client ID: 12
*****************************************
CORRUPT HEAP: multi_heap.c:187 detected at 0x3ffd7b83
abort() was called at PC 0x4008d9f6 on core 1

Backtrace: 0x4008e9b0:0x3ffd64a0 0x4008ebe1:0x3ffd64c0 0x4008d9f6:0x3ffd64e0 0x4008dfa5:0x3ffd6500 0x4008e5be:0x3ffd6520 0x400855da:0x3ffd6540 0x40087039:0x3ffd6560 0x4000bec7:0x3ffd6580 0x4015fdf1:0x3ffd65a0 0x400e515d:0x3ffd65c0 0x40194e1b:0x3ffd65e0 0x400e626f:0x3ffd6600 0x400e69c7:0x3ffd6620 0x400e6a2d:0x3ffd6640 0x400e6bb1:0x3ffd6660 0x400e6bfe:0x3ffd66a0 0x400e6c0d:0x3ffd66c0 0x400e3bae:0x3ffd66e0 0x400e3c2b:0x3ffd6710 0x400e3cac:0x3ffd6750 0x400e6aeb:0x3ffd6770 0x400e6b11:0x3ffd67a0 0x400e379d:0x3ffd67c0 0x400e3948:0x3ffd67f0 0x400e4079:0x3ffd6810 0x4008acd9:0x3ffd6840
  #0  0x4008e9b0:0x3ffd64a0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd64c0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4008d9f6:0x3ffd64e0 in multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) get_prev_free_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:187
  #3  0x4008dfa5:0x3ffd6500 in multi_heap_free_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
  #4  0x4008e5be:0x3ffd6520 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #5  0x400855da:0x3ffd6540 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #6  0x40087039:0x3ffd6560 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #7  0x4000bec7:0x3ffd6580 in ?? ??:0
  #8  0x4015fdf1:0x3ffd65a0 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #9  0x400e515d:0x3ffd65c0 in AsyncWebSocketMultiMessage::~AsyncWebSocketMultiMessage() at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #10 0x40194e1b:0x3ffd65e0 in std::_Function_handler<void (AsyncWebSocketMessage* const&), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(AsyncWebSocketMessage*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessage* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #11 0x400e626f:0x3ffd6600 in std::function<void (AsyncWebSocketMessage* const&)>::operator()(AsyncWebSocketMessage* const&) const 
at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #12 0x400e69c7:0x3ffd6620 in LinkedList<AsyncWebSocketMessage*, 
LinkedListNode>::free() at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) AsyncWebSocketClient::~AsyncWebSocketClient() at lib/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:502
  #13 0x400e6a2d:0x3ffd6640 in std::_Function_handler<void (AsyncWebSocketClient* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketClient*)#1}>::_M_invoke(std::_Any_data const&, AsyncWebSocketClient* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #14 0x400e6bb1:0x3ffd6660 in std::function<void (AsyncWebSocketClient* const&)>::operator()(AsyncWebSocketClient* const&) const at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) LinkedList<AsyncWebSocketClient*, LinkedListNode>::remove_first(std::function<bool (AsyncWebSocketClient* const&)>) at lib/ESPAsyncWebServer-master/src/StringArray.h:150
      (inlined by) AsyncWebSocket::_handleDisconnect(AsyncWebSocketClient*) at lib/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:887
  #15 0x400e6bfe:0x3ffd66a0 in AsyncWebSocketClient::_onDisconnect() at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #16 0x400e6c0d:0x3ffd66c0 in std::_Function_handler<void (void*, AsyncClient*), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*)#5}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #17 0x400e3bae:0x3ffd66e0 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
  #18 0x400e3c2b:0x3ffd6710 in AsyncClient::_close() at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
  #19 0x400e3cac:0x3ffd6750 in AsyncClient::close(bool) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
  #20 0x400e6aeb:0x3ffd6770 in AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
[... cut ...]
  #23 0x400e3948:0x3ffd67f0 in AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116   
  #24 0x400e4079:0x3ffd6810 in _async_service_task(void*) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
      (inlined by) _async_service_task at lib\AsyncTCP-master\src/AsyncTCP.cpp:197
  #25 0x4008acd9:0x3ffd6840 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

zekageri, 2020-12-10:
**This problem occours if the client wants to connect or disconnect and it is always at** `0x3ffd7b88`

```
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffd7b88
abort() was called at PC 0x4008de5b on core 1

Backtrace: 0x4008e9b0:0x3ffd8ca0 0x4008ebe1:0x3ffd8cc0 0x4008de5b:0x3ffd8ce0 0x4008e56e:0x3ffd8d00 0x400854dc:0x3ffd8d20 0x40085511:0x3ffd8d40 0x4008702d:0x3ffd8d60 0x4000beaf:0x3ffd8d80 0x4015feab:0x3ffd8da0 0x400e6409:0x3ffd8dc0 0x400e64b9:0x3ffd8de0 0x400e64e5:0x3ffd8e00 0x400e650f:0x3ffd8e20 0x400e6525:0x3ffd8e40 0x400d42cd:0x3ffd8e60 0x400d687e:0x3ffd8eb0 0x400db149:0x3ffd8fa0 0x4008acd9:0x3ffd8fc0
  #0  0x4008e9b0:0x3ffd8ca0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd8cc0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4008de5b:0x3ffd8ce0 in multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:432
  #3  0x4008e56e:0x3ffd8d00 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #4  0x400854dc:0x3ffd8d20 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #5  0x40085511:0x3ffd8d40 in heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #6  0x4008702d:0x3ffd8d60 in _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
  #7  0x4000beaf:0x3ffd8d80 in ?? ??:0
  #8  0x4015feab:0x3ffd8da0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:50
  #9  0x400e6409:0x3ffd8dc0 in LinkedList<AsyncWebSocketMessage*, LinkedListNode>::add(AsyncWebSocketMessage* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) AsyncWebSocketClient::_queueMessage(AsyncWebSocketMessage*) at lib/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:565
  #10 0x400e64b9:0x3ffd8de0 in AsyncWebSocketClient::text(AsyncWebSocketMessageBuffer*) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #11 0x400e64e5:0x3ffd8e00 in AsyncWebSocket::textAll(AsyncWebSocketMessageBuffer*) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #12 0x400e650f:0x3ffd8e20 in AsyncWebSocket::textAll(char const*, unsigned int) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #13 0x400e6525:0x3ffd8e40 in AsyncWebSocket::textAll(char*) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #14 0x400d42cd:0x3ffd8e60 in _ZL10Send_Async6StringS_$constprop$537 at lib/modbus-esp8266-master/src/Modbus.h:240        
  #15 0x400d687e:0x3ffd8eb0 in ESP_Time::Assemble_Time() at lib/modbus-esp8266-master/src/Modbus.h:240
  #16 0x400db149:0x3ffd8fa0 in ESP_Time::Tick() at lib/modbus-esp8266-master/src/Modbus.h:240
      (inlined by) Time_System(void*) at src/Own_Headers/ESP_Time.h:171
  #17 0x4008acd9:0x3ffd8fc0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

zekageri, 2020-12-10:
```
CORRUPT HEAP: Bad tail at 0x3ffdf8b8. Expected 0xbaad5678 got 0xbaad5677
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4012d8df on core 1

Backtrace: 0x4008e9b0:0x3ffd66e0 0x4008ebe1:0x3ffd6700 0x4012d8df:0x3ffd6720 0x4008e5b5:0x3ffd6750 0x400855da:0x3ffd6770 0x40087039:0x3ffd6790 0x4000bec7:0x3ffd67b0 0x401005e9:0x3ffd67d0 0x400e416c:0x3ffd6800 0x4008acd9:0x3ffd6830
  #0  0x4008e9b0:0x3ffd66e0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd6700 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4012d8df:0x3ffd6720 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x4008e5b5:0x3ffd6750 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #4  0x400855da:0x3ffd6770 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #5  0x40087039:0x3ffd6790 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #6  0x4000bec7:0x3ffd67b0 in ?? ??:0
  #7  0x401005e9:0x3ffd67d0 in esp_task_wdt_delete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:350
  #8  0x400e416c:0x3ffd6800 in _async_service_task(void*) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
  #9  0x4008acd9:0x3ffd6830 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

zekageri, 2020-12-10:
**It seems to me if i limit the connected clients to 5 in my code with a define :** `#define WEBSOCKETS_SERVER_CLIENT_MAX (5)`
**and in** `AsyncWebsocket.h` **modify the existing define:** `#define DEFAULT_MAX_WS_CLIENTS 8` to `#define DEFAULT_MAX_WS_CLIENTS 5`

zekageri, 2020-12-10:
### No it is still there :( : 

```
***********Connected Client**************
Count of Clients: 3
Client IP: 92.118.176.228
Client ID: 34
*****************************************

CORRUPT HEAP: Bad tail at 0x3ffd7a40. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4012d8df on core 1

Backtrace: 0x4008e9b0:0x3ffd6630 0x4008ebe1:0x3ffd6650 0x4012d8df:0x3ffd6670 0x4008e5b5:0x3ffd66a0 0x400855da:0x3ffd66c0 0x40087039:0x3ffd66e0 0x4000bec7:0x3ffd6700 0x4015fe89:0x3ffd6720 0x400e6355:0x3ffd6740 0x400e6b6b:0x3ffd6760 0x400e6ba5:0x3ffd6790 0x400e3831:0x3ffd67b0 0x400e39dc:0x3ffd67e0 0x400e410d:0x3ffd6800 0x4008acd9:0x3ffd6830
  #0  0x4008e9b0:0x3ffd6630 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd6650 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4012d8df:0x3ffd6670 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x4008e5b5:0x3ffd66a0 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #4  0x400855da:0x3ffd66c0 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #5  0x40087039:0x3ffd66e0 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #6  0x4000bec7:0x3ffd6700 in ?? ??:0
  #7  0x4015fe89:0x3ffd6720 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #8  0x400e6355:0x3ffd6740 in LinkedList<AsyncWebSocketMessage*, LinkedListNode>::remove(AsyncWebSocketMessage* const&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) AsyncWebSocketClient::_runQueue() at lib/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:539
  #9  0x400e6b6b:0x3ffd6760 in AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
  #10 0x400e6ba5:0x3ffd6790 in std::_Function_handler<void (void*, AsyncClient*, unsigned int, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, unsigned int, unsigned int)#4}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, unsigned int&&, AsyncClient*&&) at lib/ESPAsyncWebServer-master/src/StringArray.h:53
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #11 0x400e3831:0x3ffd67b0 in std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>::operator()(void*, AsyncClient*, unsigned int, unsigned int) const at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
      (inlined by) AsyncClient::_sent(tcp_pcb*, unsigned short) at lib\AsyncTCP-master\src/AsyncTCP.cpp:917
  #12 0x400e39dc:0x3ffd67e0 in AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116        
  #13 0x400e410d:0x3ffd6800 in _async_service_task(void*) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
      (inlined by) _async_service_task at lib\AsyncTCP-master\src/AsyncTCP.cpp:197
  #14 0x4008acd9:0x3ffd6830 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

zekageri, 2020-12-10:
**Always those Strings...**

```
CORRUPT HEAP: multi_heap.c:308 detected at 0x3ffdf788
abort() was called at PC 0x4008db6b on core 0

Backtrace: 0x4008e9b0:0x3ffd63b0 0x4008ebe1:0x3ffd63d0 0x4008db6b:0x3ffd63f0 0x4008dee5:0x3ffd6410 0x4008e56e:0x3ffd6430 0x400854dc:0x3ffd6450 0x400855e9:0x3ffd6470 0x400856ae:0x3ffd6490 0x40087049:0x3ffd64b0 0x4000bedd:0x3ffd64d0 0x400fb40f:0x3ffd64f0 0x400fb560:0x3ffd6520 0x400fb575:0x3ffd6540 0x400fb7aa:0x3ffd6560 0x400fc037:0x3ffd6580 0x400d425a:0x3ffd65a0 0x400e70c6:0x3ffd65c0 0x400e9395:0x3ffd6700 0x400e941b:0x3ffd6740 0x400e96a1:0x3ffd6780 0x400e3995:0x3ffd67a0 0x400e3a29:0x3ffd67e0 0x400e41b5:0x3ffd6800 0x4008acd9:0x3ffd6830
  #0  0x4008e9b0:0x3ffd63b0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ebe1:0x3ffd63d0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4008db6b:0x3ffd63f0 in multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) split_if_necessary at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:307
  #3  0x4008dee5:0x3ffd6410 in multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
  #4  0x4008e56e:0x3ffd6430 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
  #5  0x400854dc:0x3ffd6450 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #6  0x400855e9:0x3ffd6470 in heap_caps_realloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #7  0x400856ae:0x3ffd6490 in heap_caps_realloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #8  0x40087049:0x3ffd64b0 in _realloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:47
  #9  0x4000bedd:0x3ffd64d0 in ?? ??:0
  #10 0x400fb40f:0x3ffd64f0 in String::changeBuffer(unsigned int) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #11 0x400fb560:0x3ffd6520 in String::reserve(unsigned int) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #12 0x400fb575:0x3ffd6540 in String::copy(char const*, unsigned int) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #13 0x400fb7aa:0x3ffd6560 in String::operator=(char const*) at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #14 0x400fc037:0x3ffd6580 in String::substring(unsigned int, unsigned int) const at C:\Users\Dr.Random\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #15 0x400d425a:0x3ffd65a0 in String::substring(unsigned int) const at lib/modbus-esp8266-master/src/Modbus.h:240
  #16 0x400e70c6:0x3ffd65c0 in AsyncWebServerRequest::_parseReqHeader() at lib\ESPAsyncWebServer-master\src/WebRequest.cpp:797
  #17 0x400e9395:0x3ffd6700 in AsyncWebServerRequest::_parseLine() at lib\ESPAsyncWebServer-master\src/WebRequest.cpp:797  
  #18 0x400e941b:0x3ffd6740 in AsyncWebServerRequest::_onData(void*, unsigned int) at lib\ESPAsyncWebServer-master\src/WebRequest.cpp:797
  #19 0x400e96a1:0x3ffd6780 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at lib\ESPAsyncWebServer-master\src/WebRequest.cpp:797
      (inlined by) _M_invoke at c:\users\dr.random\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #20 0x400e3995:0x3ffd67a0 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at lib\AsyncTCP-master\src/AsyncTCP.cpp:934
  #21 0x400e3a29:0x3ffd67e0 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
  #22 0x400e41b5:0x3ffd6800 in _async_service_task(void*) at lib\AsyncTCP-master\src/AsyncTCP.cpp:1116
      (inlined by) _async_service_task at lib\AsyncTCP-master\src/AsyncTCP.cpp:197
  #23 0x4008acd9:0x3ffd6830 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

zekageri, 2021-02-26:
still websocket crash

zekageri, 2021-06-03:
[Solved with forked version from Depau](https://github.com/Depau/ESPAsyncWebServer/tree/wi-se-patches)

closed 2021-06-03 by zekageri