me-no-dev/ESPAsyncWebServer #894: Best practices for web UI/portal with device as AP

waymond91, 2020-11-25:
Hello,
Great library, great work! I have been using this to build up a small device I'm working on.

Users would connect to the network from their smartphone, and then be presented with a small web UI to help them configure their device (not just web credentials, but hardcoded parameters).

I am using LittleFS to save html/css/js files and serve them to their browser.
I've hardcoded my network credentials into the device and everything is working great.
My workflow looks something like this:

    server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/style.css", "text/css");
    });

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/index.html", String(), false, processor);
    });

    server.on("/app.js", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/app.js", "text/javascript");
    });

    server.begin();

And then any time I want to change or add a page I just route it in here and then upload it to the file system.
This all works great so far and I am happy camper.

However, In my ideal world, I'd like the device to host its own AP, and users would connect to the network, open their browser and be redirected in some memorable, convenient way to my web app.

I've had no problems hosting my own AP, but when I do so I still need to punch in the IP address - not really memorable or easy to explain to clients/customers.

What is the best way to implement this?

Is there some DNS witchcraft that would allow them to type in something like `www.my-cool-device.com` or `my-cool-device.local` and be redirected to my page?

The captive portal demo works perfectly for me, but I am confused as to how to route in my pages.
Is there a way to serve to an html page from SPIFFS/LITTLEFS for the captive portal? 
It seems that if I try something like this:

```
class CaptiveRequestHandler : public AsyncWebHandler {
public:
    CaptiveRequestHandler() {}
    virtual ~CaptiveRequestHandler() {}

    bool canHandle(AsyncWebServerRequest *request){
        //request->addInterestingHeader("ANY");
        return true;
    }

    void handleRequest(AsyncWebServerRequest *request) {
        request->send(SPIFFS, "/index.html", "text/html");
    }
};
```
My `index.html` does contain js so maybe that's not good...
```
<html lang="en">
<head>
  <title>Buddy Tracker</title>
  <meta charset="utf-8">
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/bootstrap/4.5.2/css/bootstrap.min.css">
  <link rel="stylesheet" type="text/css" href="style.css">
  <script src="https://ajax.googleapis.com/ajax/libs/jquery/3.5.1/jquery.min.js"></script>
  <script src="https://cdnjs.cloudflare.com/ajax/libs/popper.js/1.16.0/umd/popper.min.js"></script>
  <script src="https://maxcdn.bootstrapcdn.com/bootstrap/4.5.2/js/bootstrap.min.js"></script>
  <link href="https://cdn.jsdelivr.net/gh/gitbrent/bootstrap4-toggle@3.6.1/css/bootstrap4-toggle.min.css" rel="stylesheet">
  <script src="app.js"></script>
</head>
<body>

<div class="jumbotron text-center">
  <h1 class="display-5"><b>Buddy</b>Tracker</h1>
</div>
  
<div class="container">
  <div class="row">
    <div class="col-sm-12">
      <div class="card shadow-sm p-3 mb-5 rounded">
        <h3 style="text-align:center" class="text-danger">Your Safety Limits</h3>
        <h4 style="text-align:center" class="text-info">%deviceid%</h4>
        <b>Max Seperation Distance: </b>%distmax% meters<br>
        <b>Max Down Time: </b>%downmax% seconds<br>
        <b>One-Up One-Down: </b>%buddylock%<br><br>
        <button type="button" class="btn btn-danger" onclick=" goToConfig()">Change Safety Limits</button><br>
        <button type="button" class="btn btn-info" onclick=" goToBuddy()">Scan For Buddies</button>
      </div>
    </div>
  </div>
</div>

</body>
</html>
```

The captive portal no longer prompts my android device when signing into the network; and subsequently `192.164.4.1` takes me nowhere.

From [this post](https://stackoverflow.com/questions/63587623/add-multiple-spiffs-files-to-captive-portal-on-async-web-server-on-esp32) I also add the routes in `CaptiveRequestHandler` but to no avail.

```
class CaptiveRequestHandler : public AsyncWebHandler {
public:
    CaptiveRequestHandler() {
        server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/style.css", "text/css");
        });

        server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/index.html", "text/html");
        });

        server.on("/app.js", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/app.js", "text/javascript");
        });
    }
    virtual ~CaptiveRequestHandler() {}

    bool canHandle(AsyncWebServerRequest *request){
        //request->addInterestingHeader("ANY");
        return true;
    }

    void handleRequest(AsyncWebServerRequest *request) {
        AsyncResponseStream *response = request->beginResponseStream("text/html");
        response->print("<!DOCTYPE html><html><head><title>Captive Portal</title></head><body>");
        response->print("<p>This is out captive portal front page.</p>");
        response->printf("<p>You were trying to reach: http://%s%s</p>", request->host().c_str(), request->url().c_str());
        response->printf("<p>Try opening <a href='http://%s'>this link</a> instead</p>", WiFi.softAPIP().toString().c_str());
        response->print("</body></html>");
        request->send(response);
    }
};
```
The captive portal still works but I can't get to my pages.

I also tried without the captive portal, just with the SoftAP and tried to do a catch all:
```
server.onNotFound([](AsyncWebServerRequest *request) {
        request->send(SPIFFS, "/index.html", String(), false, processor);
    });
```
But this didn't work either and isn't particularly graceful...
Can anybody point me in the right direction?

I'm sorry for needing to be babied through this; I've been stumped on this part of the project for a while and I know there has to be way to do this, my approach just must be wrong.

zekageri, 2020-11-26:
### Hi! I just do it like this hope it helps

```
#include <Arduino.h>

/** VENDOR HEADERS */
#include <WiFi.h>
#include <FS.h>
#include <LITTLEFS.h> 
#include <DNSServer.h>
#include <AsyncTCP.h>
#include "ESPAsyncWebServer.h"

    DNSServer dnsServer;
    AsyncWebServer server(80);
    AsyncWebServerRequest *request;
    AsyncWebSocket ws("/CP");

    const char* ssid     = "CaptivePortalNAME";

static inline void HTTP_Requests(){
       server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/style.css", "text/css");
        });

        server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/index.html", "text/html");
        });

        server.on("/app.js", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/app.js", "text/javascript");
        });
}

class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {
    HTTP_Requests();
  }
  virtual ~CaptiveRequestHandler() {}
  bool canHandle(AsyncWebServerRequest *request){
    return true;
  }

void handleRequest(AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/Index.html.gz", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
 }

};

static const inline void Handle_Captive(){
    WiFi.softAP(ssid);
    delay(100);
    dnsServer.start(53, "*", WiFi.softAPIP());
    ws.onEvent(onWsEvent);
    server.addHandler(&ws);
    server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);
    server.begin();
    delay(100);
    IPAddress IP = WiFi.softAPIP();
    Serial.print("Started with ip: ");
    Serial.println(IP);
}

void setup() {
  Serial.begin(115200);
  Handle_Captive();
}

void loop() {
 dnsServer.processNextRequest();
}
```

**This is not complete. Just a reference / example CP**

waymond91, 2020-11-26:
Thank you for your response! I'm tinkering with the code now.
I don't understand exactly what is going on here; there is an undefined reference to a websocket event/handler `onWsEvent`.
What is the websocket supposed to be doing?

zekageri, 2020-11-26:
Oh sorry. You can comment out that line `ws.onEvent(onWsEvent);` and that `server.addHandler(&ws);` in the function `static const inline void Handle_Captive()`. Websocket is a bidirectional communication protocol to talk with the client almost seaminglessly.

zekageri, 2020-11-26:
**This will be more relevant in your case. Sorry I didn't payed enough attention to some major details.**

```
#include <Arduino.h>

/** VENDOR HEADERS */
#include <WiFi.h>
#include <FS.h>
#include <LITTLEFS.h> 
#include <DNSServer.h>
#include <AsyncTCP.h>
#include "ESPAsyncWebServer.h"

    DNSServer dnsServer;
    AsyncWebServer server(80);
    AsyncWebServerRequest *request;
    AsyncWebSocket ws("/CP");

    const char* ssid     = "CaptivePortalNAME";

static inline void HTTP_Requests(){
       server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/style.css", "text/css");
        });
        server.on("/app.js", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(SPIFFS, "/app.js", "text/javascript");
        });
}

class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {
    HTTP_Requests();
  }
  virtual ~CaptiveRequestHandler() {}
  bool canHandle(AsyncWebServerRequest *request){
    return true;
  }

void handleRequest(AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/index.html", "text/html");
    //response->addHeader("Content-Encoding", "gzip");
    request->send(response);
 }

};

static const inline void Handle_Captive(){
    WiFi.softAP(ssid);
    delay(100);
    dnsServer.start(53, "*", WiFi.softAPIP());
    //ws.onEvent(onWsEvent);
    //server.addHandler(&ws);
    server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);
    server.begin();
    delay(100);
    IPAddress IP = WiFi.softAPIP();
    Serial.print("Started with ip: ");
    Serial.println(IP);
}

void setup() {
  Serial.begin(115200);
  Handle_Captive();
}

void loop() {
 dnsServer.processNextRequest();
}
```

closed 2021-02-13 by stale[bot]

cvjmilos, 2024-03-09:
> Oh sorry. You can comment out that line `ws.onEvent(onWsEvent);` and that `server.addHandler(&ws);` in the function `static const inline void Handle_Captive()`. Websocket is a bidirectional communication protocol to talk with the client almost seaminglessly.

Sorry for bumping into this thread but this is EXACTLY what I need - websocket communication between ESP8266 and it's captive portal served page. Client connects to the ESP8266 access point, served page opens automatically upon connection and page communicates with the server, could it work or the concept of captive portal excludes websocket communication?

Belleson, 2024-03-09:
I never used ESP8266. For ESP32, web sockets work the same for AP and STA.

cvjmilos, 2024-03-09:
I cannot seem to make websockets work on the captive portal, they normally work otherwise for me. The "server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);" part interferes with the WS, I am hitting the wall here...