me-no-dev/ESPAsyncWebServer #895: Use another ESP as a web socket client

stupidVan, 2020-11-26:
Use another ESP as a web socket client
Hello.
I am a beginner in programming. I made a simple home device using your library. Thanks you.

I had a need to use a different ESP as a web socket client and listen for ws.textAll messages sent by the server.
If possible Please tell me how to start.

Sorry for the stupid question and the use of google translate

CelliesProjects, 2020-12-01:
I use this client together with ESPasyncWebServer as ws server.
 https://github.com/Links2004/arduinoWebSockets

stupidVan, 2020-12-04:
> I use this client together with ESPasyncWebServer as ws server.
> https://github.com/Links2004/arduinoWebSockets

Thank you!

closed 2020-12-04 by stupidVan

ChangeD20, 2021-03-15:
> I use this client together with ESPasyncWebServer as ws server.
> 
>  https://github.com/Links2004/arduinoWebSockets 

Hi!

I've just tried this links2004 library to connect a esp8266 as a websocketclient to my esp32 asyncwebserver  with no success. Could you please help with the configuration?

For establishing the connection I use this function as shown in the examples:

webSocket.begin("192.168.0.123", 81, "/");

I put my data in:

webSocket.begin("192.168.4.1", 80, "/");

Webserver is set up as followed:

AsyncWebServer server(80);
AsyncWebSocket ws("/ws");

Thanks in advance

ChangeD20, 2021-03-15:
> I use this client together with ESPasyncWebServer as ws server.
> https://github.com/Links2004/arduinoWebSockets

@CelliesProjects

could you help out here?

https://github.com/me-no-dev/ESPAsyncWebServer/issues/956#issuecomment-799650192