me-no-dev/ESPAsyncWebServer #898: Serve files with .gz extension to IOS ( Iphone )

zekageri, 2020-12-08:
**I'm serving my files like this:**

```
server.on("/index.html", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/index.html.gz", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

_It is working an all operating systems except Iphones. Iphone tries to download the index.html instead of serving it.
If i delete the .gz from the file name it is working properly, but i don't want to delete it in every file alwys if i modify and zip some of them._ 
**Any workaround on this?**

ismal, 2020-12-08:
The workaround is to rename those .gz into something that IOS will not understand. So I am serving .js.gz as .jsgz and it works well exactly in a way that you described above. In your case you can try to use "LITTLEFS, "/index.htmlgz",

zekageri, 2020-12-08:
Yeah i'm currently serving the files without the .gz in the filename. 
When i zipping the files i always delete the " .gz " from the end of the file names (for safari). 
But it is really frustrating.

lorol, 2020-12-08:
Unfortunately, you have to cope with this, 
Depends of how you prepare the gzipped files, you can use a script / batch file etc to compress then rename it before upload as you wish.

zekageri, 2020-12-09:
Oh nice. I didn't thought about a batch script. I will investigate it.
Thank you!

ismal, 2020-12-09:
another simple script for renaming can look like this:
for f in *.html.gz; do     mv -- "$f" "${f%.html.gz}.html"; done

zekageri, 2020-12-09:
Thank you all! I will close this issue and open it on some Apple forum!

closed 2020-12-09 by zekageri