me-no-dev/ESPAsyncWebServer #899: Workarounds to make SSL/TLS work

tom-ch1, 2020-12-09:
Setting-up an SSL/TLS enabled Server using the ESPAsyncWebServer library is not yet fully documented and straight-forward. I spent a lot of time and finally got it to work. I'd like to share what I have learned, so you can make it work, too.

## My setup and basic requirements
* Sonoff-basic (ESP8266)
* Arduino IDE with debugging enabled
* Create a sketch which implements a https Server allowing upload of an updated sketch
* Install [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) and its dependency [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) manually into the Arduino IDE
* starting point is the example sketch [Setting up the Server](https://github.com/me-no-dev/ESPAsyncWebServer#setting-up-the-server)
* Adding SSL functionality according to #75 

## Here's an account of the caveats and how to work around them
1. Compile error "class AsyncServer has no member named beginSecure" 
    * This is #392: Solution: `#define ASYNC_TCP_SSL_ENABLED 1` **before** `#include <ESPAsyncTCP.h>`
1. Link error: `no matching function for call to AsyncClient::_recv(tcp_pcb*&, pbuf*&, int)` or similar
    * This is a bug: https://github.com/me-no-dev/ESPAsyncTCP/issues/131
    * Fix: Install the patched Version of ESPAsyncTCP: https://github.com/jeroenst/ESPAsyncTCP
1. Link error: `undefined reference to AsyncWebServer::onSslFileRequest(std::function<int (void*, char const*, unsigned char**)>, void*)` or similar
    * This is a bug: See #753 or https://gitter.im/me-no-dev/ESPAsyncWebServer?at=5d746080c59390272030908c 
    * add `#define ASYNC_TCP_SSL_ENABLED 1` to the top of `libraries/ESPAsyncTCP/src/async_config.h` (Arduino IDE)
    * This will increase code size even for projects not using SSL, but I didn't find another solution for it.
    * don't add it as a compiler flag in platform.local.txt (Arduino IDE), it won't work!
1. Now it compiles, but you may get a runtime error: `Error: Feature not supported` after calling `server.beginSecure()`
    * I tracked that one down to https://github.com/igrr/axtls-8266.
    * I thought it just didn't work because I used an unsupported cypher suite. But it wasn't that and I couldn't figure out the reason
    * but I found a configuration which works:
        * `openssl genrsa -out Key.pem 1024`
        * `openssl req -x509 -out Cert.pem -key Key.pem -new -sha256 -subj /CN=your.domain -addext "keyUsage=digitalSignature,keyEncipherment" -addext extendedKeyUsage=serverAuth`

I hope I could help somebody trying to set up an SSL-protected update-server

zekageri, 2020-12-10:
Can it work with esp32 using AsyncTCP?

jcassel, 2020-12-16:
@tom-ch1 Thank you for your work here in navigating the issues and then giving us all this leg up. It has really saved me a lot of time. Much appreciated. 

One additional item that had me scratching my head for a few min was that once you have the files in place and the call to onSSLFileRequest you also need to add in the SPIFFS.Begin() call before you start the server.

tom-ch1, 2020-12-18:
@zekageri : I don't know, but I have come across some sites which state that the esp32 doesn't have enough resources for SSL.

@jcassel : Thanks for pointing that out. In fact, my version is using LittleFS.

For a complete sketch see my example at https://gitlab.users.ch.eu.org:10443/smarthome/eaws-ssl-updater

closed 2021-03-05 by stale[bot]

vslinuxdotnet, 2021-07-15:
Just as Note: In the the SDK 3.X don't work because the openssl libs was removed.

mhaberler, 2021-11-09:
> @zekageri : I don't know, but I have come across some sites which state that the esp32 doesn't have enough resources for SSL.
>
> @jcassel : Thanks for pointing that out. In fact, my version is using [...]

unfortunately that site went away - any replacement location?

Puding07, 2021-11-29:
> unfortunately that site went away - any replacement location?

@mhaberler I made one example WebServer sketch for ESP8266 using TLS. Check it out at: https://gitlab.com/artonworkstm/asyncwebservertls

The only thing I miss is session cache which would fasten up a bit the requests. I'll dig myself into it in the next days.

mhaberler, 2021-11-29:
> @mhaberler I made one example WebServer sketch for ESP8266 using TLS. Check it out at: https://gitlab.com/artonworkstm/asyncwebservertls
> 
> The only thing I miss is session cache which would fasten up a bit the requests. I'll dig myself into it in the next days.

thanks, appreciated!

tom-ch1, 2021-12-05:
> unfortunately that site went away - any replacement location?

Yes, I put my Gitlab site behind a reverse proxy, so it's easier to find: https://gitlab.users.ch.eu.org/smarthome/eaws-ssl-updater

ZanzyTHEbar, 2022-11-08:
> @zekageri : I don't know, but I have come across some sites which state that the esp32 doesn't have enough resources for SSL.
>
> @jcassel : Thanks for pointing that out. In fact, my version is using [...]

that doesn't even make sense ... the ESP32 has a built-in SSL webserver header, it has an on-board cryptocell chip and has oodles more resources than the ESP8266. So, that's definitely not true.

TheMiNuS, 2023-02-18:
I confirm it works running HTTPS server on an ESP8266 with this patch.
Is there a plan to integrate that code to the main branch ?

uGiFarukh, 2023-02-21:
Any way this can be used for the ESP32 using AsyncTCP library?

danielaac83, 2023-11-15:
I join to question, SSL in ESP32 winth library ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer is possible?
Thanks

zekageri, 2023-11-15:
Unfortunately no