me-no-dev/ESPAsyncWebServer #901: How do I return a different http result  (500) on upload?

mark-hahn, 2020-12-14:
The only way I know how to upload a file uses this standard code ...
``` server.on("/upload", HTTP_POST, 
            [](AsyncWebServerRequest *request) {
              request->send(200);
            }, onUpload);
```
But in some situations, like running out of file system space in onUpload, I need to return a different code than 200.  How do I do this?

mark-hahn, 2020-12-16:
Anyone?  I'm still stuck on this since I can't find any examples of handling errors on uploads.  I'm kind of baffled that so much code doesn;t foillow the practice of always handling errors.

BlueAndi, 2020-12-16:
You can set a global error flag in the onUpload() and consider it in the lambda function.

mark-hahn, 2020-12-16:
Thanks!   But I'm not sure how I'm going to handle simultaneous uploads.  The web app doing the upload can be run by multiple users.  The odds of two uploading at once is unlikely, but it could happen.

themindfactory, 2020-12-16:
Spawn a new task and pass it the request is what I did some years ago for
this scenario.

mark-hahn, 2020-12-16:
I don't know how to spawn a task in arduino.  I guess I could make a fake task using the main loop.

Do you mean to spawn it in the lambda function or in the handler?  I'm not sure how the task could pass the handler results to the lambda function before it finishes.

The whole concept of having both the lambda function and the handler is kind of a bizzare architecture, a duplication of responsibilities.  Can I have an empty lambda and just call `request->send` in the last call to the handler?  I seem to remember trying this.

themindfactory, 2020-12-16:
I'd have to find my old code to see what I did..

lorol, 2020-12-16:
BTW, running out of FS space is also challenging itself, you may not be able to get this before it happens. I believe because of buffering etc.

mark-hahn, 2020-12-16:
I used full fs as an example.  Wouldn't a full fs give a write error (zero
bytes written)?

lorol, 2020-12-16:
Sorry what is full fs?
See here something related - tests to figure out sectors / files available at different fs.
https://github.com/lorol/LITTLEFS/issues/12#issuecomment-730430141

mark-hahn, 2020-12-16:
> lorol/LITTLEFS#12 (comment)

By "full fs" I meant a fs state where no more files of the size I'm writing
can be written.   I can't write a gigabyte :-)

lorol, 2020-12-16:
@mark-hahn  Sorry for my ignorance.
An example please?

BlueAndi, 2020-12-16:
Why you don't use the AsyncWebServerRequest informations? It contains the info about the client, which requested the upload. You could combine this with error information and consider it in the lambda function.

lorol, 2020-12-16:
Or maybe check  (SPIFFS.totalBytes() - SPIFFS.usedBytes()) and return 500 before uploading . I didn't have patience to try,  it may work

mark-hahn, 2020-12-16:
> @mark-hahn Sorry for my ignorance.
> An example please? What exactly do you expect to return you the 0 (or whatever is criteria for full)?

```
if(file.write(data, len) == 0) fileSystemIsFull();
```

lorol, 2020-12-16:
Thanks. Unfortunately, getting "near" the capacity file.write() can return you >0 but file is not written.

mark-hahn, 2020-12-17:
That is a nasty bug.  Thanks for the heads-up.

lorol, 2020-12-17:
It depends on FS and file size / sectors fit,  you can try yourself. With SPIFFS, it looks close to what you expect. Ffat - I didn't try to much. On newest LittleFS, you need some "room" before the end of partition.

closed 2021-03-05 by stale[bot]