me-no-dev/ESPAsyncWebServer #902: no matching function for call to 'esp8266webserver: :ESP8266WebServerTemp1ate<WiFiServer>: :on(const char [19], HTTPMethod, setup(): :_lambdal) ' candidates are:

MisiekDP, 2020-12-15:
Hey, I have a problem with this library. Today I wanted to program an project with getting URL with parameters but all the time I got an errors. I use a PlatformioIO too.
There are errors:

[image]

[image]

[image]

[image]

```
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266WebServer.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <ESPAsyncTCP.h>
#include <ESP8266WiFi.h>
// #include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

// Pins GPIO
int ledRed = 0;
int ledYellow = 2;
int buttonRed = 5;
int buttonYellow = 16;
DHT dht(4, DHT11);

// WiFi config
IPAddress ip(10, 6, 0, 120);        // Static IP adress
IPAddress gateway(10, 6, 0, 5);     // Gateway
IPAddress subnet(255, 255, 0, 0);   // IP maska
ESP8266WebServer server(80);        // Server port

...

void setup() {
  Serial.begin(9600);

 ...

  // Starting the server and assigning actions
  server.on("/", websiteLoad);
  server.on("/w1t", websiteLed1Toggle);
  server.on("/w2t", websiteLed2Toggle);
  server.on("/sendResponseIfESP8266Works", sendResponseIfESP8266Works);
  server.on("/sendData", sendData);
  server.on("/led1toggle", led1toggle);
  server.on("/led2toggle", led2toggle);
  // server.on("/seek_bar_led2_pwm", seekBarLed2PWM);
  // respond to GET requests on URL /heap

  server.on("/seek_bar_led2_pwm", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebParameter* p = request->getParam(0);
    Serial.print("Param name: ");
    Serial.println(p->name());
    Serial.print("Param value: ");
    Serial.println(p->value());
    Serial.println("------");

    // server.send(200);
    request->send(200, "text/plain", "message received");
  });

  server.begin();
}

void loop() {
  // Waiting for client connecting
  server.handleClient();
...
}
```

egrekov, 2020-12-18:
@MisiekDP `#define WEBSERVER_H`, will solve the compilation problem, but if you use constants like `HTTP_POST`, `HTTP_PUT`, 
then they will not work properly. [issue904](https://github.com/me-no-dev/ESPAsyncWebServer/issues/904)
```c++
#define WEBSERVER_H
#include "ESPAsyncWebServer.h"
```

MisiekDP, 2020-12-18:
> @MisiekDP `#define WEBSERVER_H`, will solve the compilation problem, but if you use constants like `HTTP_POST`, `HTTP_PUT`,
> then they will not work properly. [issue904](https://github.com/me-no-dev/ESPAsyncWebServer/issues/904)
> 
> ```c++
> #define WEBSERVER_H
> #include "ESPAsyncWebServer.h"
> ```

Ok, thanks for reply :D But now I can't test it :/

wiseda51, 2021-01-22:
Just starting to use AsyncWebServer and  browsing through issues... You are crossing classes between old ESP8266WebServer (synchronous server run in loop) and AsyncWebServer (new ESP32/ESP8266 async webserver).  So get rid of ESP8266WebServer.h and change your object to AsyncWebServer server(80);.

closed 2021-06-02 by stale[bot]