me-no-dev/ESPAsyncWebServer #903: OTA update of compressed SPIFFS (ESP32)

gilphilbert, 2020-12-16:
I'm working on a project where I'm leveraging SPIFFS to serve a static website with an Event Source, all from ESPAsyncWebServer on an ESP32. Many of the final devices may not be connected to the internet, so I'm looking at OTA for updates but I want a single OTA that updates the firmware and the web content in the SPIFFS partition.

To get there, I've built a system that updates SPIFFS OTA (using the sample code in this repo). The SPIFFS update _may_ include a new `firmware.bin` file. When the ESP32 resets after an OTA update, it will check for the firmware.bin file in SPIFFS - if it finds one, it updates the firmware and deletes the file then resets. This allows me to have a single file to handle both firmware and SPIFFS updates, much easier for the user. I couldn't think of an easier way to get to this result.

  However, the SPIFFS partition is now fairly large in order to contain the firmware and site files, so now my spiffs.bin file is pretty big and takes a few minutes to upload over WiFi. I looked online and it seems that the newer ESP32 code can detect and handle compressed firmware files, but there's no mention of SPIFFS. I compressed (`gzip -9 spiffs.bin`) the SPIFFS file and uploaded but my SPIFFS partition was just garbage, so I'm guessing that doesn't work.

Did I do something wrong, or is this not supported (yet... 😄)

avillacis, 2020-12-19:
You could take a look at our web framework for ESP32, [YUBOX-Now](https://github.com/yubox-node-org/yubox-now). I wrote a web upload-based OTA capability with both firmware file and SPIFFS content packed in a tar.gz file. On upload, the gzip data is decompressed on the fly and the firmware file is never stored on the filesystem, but instead directly written to the partition.

gilphilbert, 2020-12-23:
Thanks, I'll take a look and see about porting it to my application.

I'm surprised that the ESP32 can't do this natively since the chip presumably supports compression of some sort since esptool.py compresses images before it sends them over serial. I guess that wasn't ported outside of serial programming.

Anyway, thanks for the hints!

closed 2020-12-23 by gilphilbert