me-no-dev/ESPAsyncWebServer #906: Upload file to ch376 from espasyncwebserver POST

petervanderwalt, 2020-12-18:
I know I am being stupid but I just can't figure this one out (:

I have a sketch running AsyncWebserver, and a Flashdrive accessed using library https://github.com/djuseeq/Ch376msc

I got it working to the point where I can POST a large file (say 2mb) to the sketch, the handler create the file, and start writing data.

```
void handleUsbUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    Serial.printf("UploadStart: %s\n", filename.c_str());
    // Custom filenames causes issues opening file
    String fileName = filename.substring(0, 8);
    fileName = fileName += ".EXT";
    fileName.toUpperCase();
    flashDrive.setFileName(fileName.c_str());  
    flashDrive.openFile();
    uploadedBytes = 0;
  }

  flashDrive.writeFile((char*)data, len);
  //flashDrive.writeRaw(data, len);

  uploadedBytes = uploadedBytes+len;
  Serial.print("Recv: ");
  Serial.print(len);
  Serial.print(" in chunk, total bytes recv: ");
  Serial.println(uploadedBytes);

  if(final){
    delay(1000);
    flashDrive.closeFile(); 
    uploadedBytes = 0;
    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index+len);
  }
}
```

The handler received the file in chunks: (MTU size, understandable)

```
Recv: 1436 in chunk, total bytes recv: 235339
Recv: 1436 in chunk, total bytes recv: 236775
Recv: 1436 in chunk, total bytes recv: 238211
Recv: 1436 in chunk, total bytes recv: 239647
Recv: 280 in chunk, total bytes recv: 239927
```

But if I look at the data written to disk afterwards, it skipped a lot of the data in the flashdrive.writefile (tried writeraw too)

It writes the first part of the chunk, rest doesn't show up in resultant file. When next chunk arrives it writes the first part of it again, throwing away the rest
What I am now not sure of is:

 * Is the uint8_t data buffer 1436 bytes? Am I correct to cast it to char to make https://github.com/djuseeq/Ch376msc/blob/b280b1842b2688316bfffc7f6d968d7dfab599f4/src/Write.cpp#L12-L14 work?
 * In case casting is the issue, I tried writeRaw of the uint8_t *data buffer directly, as per https://github.com/djuseeq/Ch376msc/blob/b280b1842b2688316bfffc7f6d968d7dfab599f4/src/Write.cpp#L16-L18 - same result
 * Is there some limit on how big a buffer https://github.com/djuseeq/Ch376msc can write in one call? Should I break up the buffer into chunks?
 * I also tried writing one character at a time https://github.com/djuseeq/Ch376msc/blob/b280b1842b2688316bfffc7f6d968d7dfab599f4/src/Write.cpp#L20-L22 (for loop over the uint8_t data, write for each data[i]) but that takes too long and triggers the WDT

lorol, 2020-12-18:
A (very ugly) workaround:
https://github.com/lorol/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L264
https://github.com/lorol/ESPAsyncWebServer/blob/master/src/edit.htm#L215

petervanderwalt, 2020-12-18:
> A (very ugly) workaround:
> https://github.com/lorol/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L264
> https://github.com/lorol/ESPAsyncWebServer/blob/master/src/edit.htm#L215

(: Sorry I am not sure what I am looking at, can you elaborate, why is there a workaround, is this some kind of bug in me-no-dev/ESPAsyncWebServer?

lorol, 2020-12-18:
I couldn't say it is a bug but definitely a limitation.
I faced similar trouble in the past and needed to split a post upload to chunks. In my case, it was a need to upload of text data back (the edited file) because an old iOS device didn't like PUT method. I couldn't find a nicer way to achieve this.

petervanderwalt, 2020-12-18:
Figured it out

CH376 library can only take 256 bytes at a time:

```

closed 2020-12-18 by petervanderwalt