me-no-dev/ESPAsyncWebServer #907: I can't make Captive Portal Work Properly

Nathan-ma, 2020-12-19:
I was following the Captive Portal example and it works great!

But I can only access what is inside the handler, all the methods based on server.on gets inaccessible.

[image]

How can I work with Handlers and captive portal and yet have all the server.on functions working?

BlueAndi, 2020-12-19:
Register the CaptiveRequestHandler after registering stuff with on() and try again.

Nathan-ma, 2020-12-19:
Thanks, will try

Nathan-ma, 2020-12-19:
> Register the CaptiveRequestHandler after registering stuff with on() and try again.

BlueAndi! 

it is working now, Thanks!

All I need to figure out is how to redirect to the index page since request->redirect("/") doesn't seem to be working, but i can figure this out!

closed 2021-02-13 by Nathan-ma