me-no-dev/ESPAsyncWebServer #909: websocket flush, or detect all sent?

mark-hahn, 2020-12-23:
I'm sending a series of websocket packets to the browser using ws.binary().  These are progress events for http update.  After the last is sent I reboot with ESP.restart().  

I'm having a problem where the restart happens before the last progress reports get to the browser and the browser hangs waiting.  I've set a delay of 5 seconds before the restart but even then sometimes the last is still not sent.  Even 5 seconds is a long extra delay for the user.

Is there some way to wait for the packets queued up to be sent?  Sort of like a flush command?

lorol, 2020-12-23:
```
server.on("/hw-reset", HTTP_GET, [](AsyncWebServerRequest *request){
      request->onDisconnect([](){
#ifdef ESP32
        ESP.restart();
#elif defined(ESP8266)
        ESP.reset();
#endif
      });
    request->send(200, "text/plain","Restarting ...");
  });
```

mark-hahn, 2020-12-23:
Duh!  Thanks mucho.

lorol, 2020-12-23:
Not sure if this will be useful as it is at http side, not on ws event. But you can adapt and try.  In works case, let the browser request reboot through http not by websocket

mark-hahn, 2020-12-23:
I set up a reset request by websocket.  My websocket appears to be dying though during the update, which may have been the real problem.  I'll try using your http example.

Why does httpupdate block?  That is a pita.

closed 2021-03-19 by stale[bot]

djbeppe, 2022-07-10:
Does anyone experienced esp8266 always crushes after receiving about 251-255 websocket events? May it be websockets* buffer (array) is limited to 256 items? is it possible to flush the buffer programmatically to prevent crush?
Thanks in advance :-)