me-no-dev/ESPAsyncWebServer #910: WebSocket null termination crashes ESP8266/8285 or ESP32

myxiplx, 2020-12-24:
Hi,

I'm no Arduino expert, but I'm finding that when working with WebSockets the example in the documentation will crash ESP8266 based systems:
https://github.com/me-no-dev/ESPAsyncWebServer#async-websocket-plugin

On ESP32, the example in the documentation works fine, on this platform when WS_TEXT data is received the example adds nulls manually to terminate incoming text data with the line:
`data[len] = 0;`

However, on the ESP8266 or ESP8285 this line causes crashes after just a few WebSockets are received.  Removing the line solves the crash problem on the 8266, but then causes the code to crash on the ESP32.  So to get around this I'm having to wrap this line in a #ifdef:

```
#ifdef ESP32
  data[len] = 0;  //ESP32 requires null termination to be added.  ESP8266 crashes if we have this line
#endif
```

Is this a bug somewhere in the WebSocket layer?

BlueAndi, 2020-12-27:
IMHO the bug is in the example. It assumes that the size of the given buffer is always (len + 1) byte, but this is wrong!
Copy them byte for byte to e.g. String and you are fine.

myxiplx, 2021-02-18:
Adding for completeness in case any other beginners run into this.  I solved the problem by having a separate buffer to receive commands, which also allows me to handle fragmented commands that have been split across multiple sockets.

When receiving data I now copy (or append) the data received to a buffer, and null terminate the buffer (which I've already checked is long enough for the command, plus the null termination).

if (len < Global::MAX_COMMAND_SIZE) {
   strncpy((char*)gCommandBuffer, (char*)data, len); 
   gCommandLength = len;
   gCommandBuffer[gCommandLength] = 0;  //Ensure null terminated      
}

closed 2021-06-16 by stale[bot]