me-no-dev/ESPAsyncWebServer #911: I cannot use process function

oguzkaganyaglioglu, 2020-12-26:
When I compile the code I get this error. What am I doing wrong? How can I fix it?

25:67 is the bold word "**this**"

> src\ServerManager.cpp: In lambda function:
src\ServerManager.cpp:25:67: error: 'this' was not captured for this lambda function
         request->send(LittleFS, "/status.html", String(), false, [**this**](const String &var) -> String { return statusProcessor(var); });

*** [.pio\build\nodemcuv2\src\ServerManager.cpp.o] Error 1

in .cpp file
```
void MyServerManager::init()
{
    if (!LittleFS.begin())
    {
        Serial.println("Something went wrong while begin LittleFS");
    }

    server->on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        request->send(LittleFS, "/status.html", String(), false, [this](const String &var) -> String { return statusProcessor(var); });
    });
    server->begin();
}
```

in .h file

```
#ifndef ServerManager_h
#define ServerManager_h
#include <Arduino.h>
#include <ESPAsyncWebServer.h>

class MyServerManager
{
private:
    AsyncWebServer *server = NULL;
    AsyncWebSocket *ws = NULL;
    void notifyClients(const String *msg);
    void handleWebSocketMessage(void *arg, uint8_t *data, size_t len);
    void onEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len);
    void initWebSocket();
    String statusProcessor(const String &var);

public:
    MyServerManager(uint16_t port, const String url);
    ~MyServerManager();

    void init();
};

#endif
```

BlueAndi, 2020-12-26:
Why do yoou use a lambda function form the statusProcessor, instead of using it directly in the on()? There is btw your problem,  because you try to capture this in the 2nd lambda, but it does not exists there.

oguzkaganyaglioglu, 2020-12-27:
Because when I use it like that I get another error.

```
    server->on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        request->send(LittleFS, "/status.html", String(), false, statusProcessor);
    });
```

Error: 

> src\ServerManager.cpp:25:81: error: no matching function for call to 'AsyncWebServerRequest::send(fs::FS&, const char [13], String, bool, <unresolved overloaded function type>)'

oguzkaganyaglioglu, 2020-12-27:
When I add this to the outer lambda function it worked. So is there another way to do it without seccond lambda?

this is worked:
```
    server->on("/", HTTP_GET, [this](AsyncWebServerRequest *request) {
        request->send(LittleFS, "/status.html", String(), false, [this](const String &var) -> String { return statusProcessor(var); });
    });
```

BlueAndi, 2020-12-27:
Now I see, the send method expects a function, not a method. That is why without the 2nd lambda, it won't work. But your last proposal do.

closed 2020-12-27 by oguzkaganyaglioglu