me-no-dev/ESPAsyncWebServer #912: LittleFS & ESPAsyncWifiManager

rogerknop, 2020-12-31:
Hi

I am trying to use the Async WifiManager and the LittleFS in parallel to have separated files for .html, .css and .js.
The Async WifiManager starts and I can setup my network connection with a static IP.
But as soon as I upload new html, css and js files with the LittleFS file manager, the configuration is lost and I have to setup my network configuration again.

I would highly appreciate some hints how to solve that.

Thanks in advance,
Roger

lorol, 2020-12-31:
Hi, what is "LittleFS file manager" - can you share your code?
Assuming you are using esp32 and the LITTLEFS library I published,  the problem is likely because LittleFS does not create folders when you ask a non existing new file to be created.
You can try with value =1 here:
https://github.com/lorol/LITTLEFS/blob/master/src/esp_littlefs.c#L37

rogerknop, 2020-12-31:
LITTLEFS correct.
And the file upload tool for Arduino IDE.

`************************************** WebServerTemplate.ino **************************************

#include <LittleFS.h>

#include <RokWifi.h>

#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

String processor(const String& var){
  Serial.println("process variable " + var);
  if(var == "TITLE"){
    return "ESP - Template";
  }
  return String();
}

void setup() {
  Serial.begin(115200);
  if (LittleFS.begin()) {
    Serial.println("Filesystem mounted");
  }
  else {
    Serial.println("ERROR!!! Filesystem not mounted...");
  }
  initRokWifi(&server);
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(LittleFS, "/index.html", String(), false, processor);
  });
  server.begin();
}

void loop() {}

************************************** RokWifi.h **************************************
#ifndef __ROKWIFI__
#define __ROKWIFI__
#include "Arduino.h"
#include <ESPAsyncWebServer.h>     //Local WebServer used to serve the configuration portal
extern void initRokWifi(AsyncWebServer *importserver);
#endif

************************************** RokWifi.cpp **************************************
#include "RokWifi.h"
#include <LittleFS.h>                   //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

#include <ESPAsyncWebServer.h>     //Local WebServer used to serve the configuration portal
#include <ESPAsyncWiFiManager.h>       

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include "Arduino.h"

char _static_ip[16] = "192.168.1.";
char _static_gw[16] = "192.168.1.1";
char _static_sn[16] = "255.255.255.0";
bool _shouldSaveConfig = false;

DNSServer dns;

void _saveConfigCallback () {
  Serial.println("Should save config");
  _shouldSaveConfig = true;
}

void initRokWifi(AsyncWebServer *server) {
  if (LittleFS.exists("/config.json")) {
      Serial.println("reading config file");
      File configFile = LittleFS.open("/config.json", "r");
      if (configFile) {
          Serial.println("opened config file");
          size_t size = configFile.size();
          // Allocate a buffer to store contents of the file.
          std::unique_ptr<char[]> buf(new char[size]);
          configFile.readBytes(buf.get(), size);
          DynamicJsonDocument json(1024);
          auto deserializeError = deserializeJson(json, buf.get());
          serializeJson(json, Serial);
          if ( ! deserializeError ) {
              if (json["ip"]) {
                  Serial.println("setting custom ip from config");
                  strcpy(_static_ip, json["ip"]);
                  strcpy(_static_gw, json["gateway"]);
                  strcpy(_static_sn, json["subnet"]);
                  Serial.println(_static_ip);
              } else {
              Serial.println("no custom ip in config");
              }
          } else {
              Serial.println("failed to load json config");
          }
      }
  }
  AsyncWiFiManager wifiManager(server, &dns);

  wifiManager.setSaveConfigCallback(_saveConfigCallback);

  //set static ip
  IPAddress _ip, _gw, _sn;
  _ip.fromString(_static_ip);
  _gw.fromString(_static_gw);
  _sn.fromString(_static_sn);

  wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);

  wifiManager.setMinimumSignalQuality();

  if (!wifiManager.autoConnect("ESPWifiSetup", "")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //save the custom parameters to FS
  if (_shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonDocument json(1024);
    json["ip"] = WiFi.localIP().toString();
    json["gateway"] = WiFi.gatewayIP().toString();
    json["subnet"] = WiFi.subnetMask().toString();
    File configFile = LittleFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    serializeJson(json, Serial);
    serializeJson(json, configFile);
    configFile.close();
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  Serial.println(WiFi.gatewayIP());
  Serial.println(WiFi.subnetMask());
}
`

lorol, 2020-12-31:
@rogerknop 
Sorry, I cannot help you - this is esp8266 project - it has different LittleFS implementation.
See https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS
Maybe @earlephilhower can help you

rogerknop, 2021-01-01:
Hi lorol,
wish you a Happy New Year!
I am confused now, because I am using an ESP8266.
With your link I found also a different upload tool: ESP32 Sketch Data upload versus ESP8266 LittleFS Data upload.
I cannot find a hint, which one I should use?
In your example you are using SPIFFS.
My assumption was that this will be deprecated. Therefore I used LittleFS.
Regards, Roger

lorol, 2021-01-01:
Hi @rogerknop 
Happy New year, too!
Sorry for the confuse.
Just disregard my previous messages. ESP32 is completely different platform than ESP8266.
Use only the ESP8266 tools for your case

wiseda51, 2021-01-22:
If I'm interpreting your issue correctly you are replacing the entire file system when you update your files, this is the same as formatting a file system; the previous contents are lost.  I believe the tool simply takes a file-system image and writes it to the appropriate esp partition (e.g. "spiffs").  So this is working as designed and not really a problem.  You can download individual files via the SPIFFSEditor (I'm not aware of any example which supports multiple files but conceptually it's not too difficult).

rogerknop, 2021-01-25:
Yes that was also the outcome of my research :-)
I created a default config file for development as workaround.
Thanks, Roger

closed 2021-01-25 by rogerknop

louis-chevallier, 2024-10-04:
your comment 
    //this needs to be first, or it all crashes and burns...
helped me ... is there any rationale behind ???