me-no-dev/ESPAsyncWebServer #916: ESP_AsyncFSBrowser doesn't browse FS???

mike-s123, 2021-01-03:
ESP_AsyncFSBrowser seems to be an example for websockets? Presumably, it was meant to be the Async equivalent of the ESPWebServer [FSBrowser example](https://github.com/esp8266/ESPWebServer/tree/master/examples/FSBrowser). There's even stuff there (like ace.js in the data directory) which seems to support that.

wiseda51, 2021-01-22:
The (static) file serving is just the one line:
server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.htm");
Everything else is just showing other things which can be done and aren't needed for simply serving files (like the websockets code).  The data directory is for SPIFFSEditor files used by: 
#ifdef ESP32
  server.addHandler(new SPIFFSEditor(MYFS, http_username,http_password));
#elif defined(ESP8266)
  server.addHandler(new SPIFFSEditor(http_username,http_password, MYFS));
#endif

CosmicMac, 2021-04-06:
Use `/edit` URL to open the editor (this should probably be mentioned in the example)

closed 2021-06-22 by stale[bot]