me-no-dev/ESPAsyncWebServer #918: Page load wait forever frequently for file load!

Speedy2k, 2021-01-11:
I have a ESP32 serving a bootstrap UI with some JS file. When i load the page, it get stuck loading pretty often.

Here is how i init the server:
```
  server.serveStatic("/advanced.html", SPIFFS, "/advanced.html").setAuthentication((char*)loginUser.c_str(), (char*)loginPass.c_str());
  server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");
  server.serveStatic("/css/", SPIFFS, "/css/");
  server.serveStatic("/js/", SPIFFS, "/js/");
  server.serveStatic("/img/", SPIFFS, "/img/");
  //server.serveStatic("/", SPIFFS, "/");
  
  server.on("/logout", HTTP_GET, [](AsyncWebServerRequest * request){
    request->requestAuthentication();
    request->redirect("/");
  });
  server.onNotFound(onPageNotFound);
  server.addHandler(&ws);
  server.begin();
}
// Callback: send 404 if requested file does not exist
void onPageNotFound(AsyncWebServerRequest *request) {
  IPAddress remote_ip = request->client()->remoteIP();
  Serial.println("[" + remote_ip.toString() +
                  "] HTTP GET request of " + request->url());
  request->send(404, "text/plain", "Not found");
}
```

And here is my HTML:

```
<!DOCTYPE html>
<html lang="en-US">
	<head>
		<meta charset="UTF-8">
		<meta name="viewport" content="width=device-width, initial-scale=1">
		<meta http-equiv="X-UA-Compatible" content="IE=edge">
		<meta name="apple-mobile-web-app-capable" content="yes">
		<meta name="apple-mobile-web-app-status-bar-style" content="black">
		<meta name="apple-mobile-web-app-title" content="ErabliTEK Inc.">
		<meta name="viewport" content="width=device-width, minimal-ui, initial-scale=1.0, user-scalable=no">
		<title id="title">wifi</title>
		<link rel="icon" href="/img/favicon.png">
		<link rel="stylesheet" type="text/css" href="/css/bootstrap.min.css">
	</head> 
	<body>
		
		<nav class="navbar navbar-expand-lg navbar-light bg-light">
			<div class="container">
			  <a class="navbar-brand" href="#">
				<img src="/img/navbar.png" height="25px" alt="">
			  </a>
			  <button class="navbar-toggler" type="button" data-toggle="collapse" data-target="#navbarText" aria-controls="navbarText" aria-expanded="false" aria-label="Toggle navigation">
				<span class="navbar-toggler-icon"></span>
			  </button>
			  <div class="collapse navbar-collapse" id="navbarText">
				<ul class="navbar-nav mr-auto">
					<li class="nav-item">
					  <a class="nav-link" href="/">Main </a>
					</li>
					<li class="nav-item">
					  <a class="nav-link" href="/info.html">Info</a>
					</li>
					<li class="nav-item active">
					  <a class="nav-link" href="/wifi.html">Wifi</a>
					</li>
					<li class="nav-item dropdown">
						<a class="nav-link dropdown-toggle" href="#" id="configDropdown" role="button" data-toggle="dropdown" aria-haspopup="true" aria-expanded="false">
							Config
						</a>
						<div class="dropdown-menu" aria-labelledby="navbarDropdown">
							<a class="dropdown-item" href="/config.html">User</a>
							<a class="dropdown-item" href="/advanced.html">Tech</a>
						</div>
					</li>
				</ul>
				<span class="navbar-text" id="navSerial"></span>
			  </div>
			</div>
		</nav>
		
		<div class="container" style="margin-top:10px">
			<div class="row">
				<div class="col-sm-12">
					<div class="card">
						<div class="card-header">
							<h4>Wifi Configuration.</h4>
						</div>
						<div class="card-body">
							<form>
								<div class="row">
									<div class="col-12 col-lg-6 align-self-center">
										<label>Connected to:</label>
									</div>
									<div class="col-12 col-lg-6 align-self-center">
										<label>Signal:</label>
									</div>
								</div>
								<div class="row">
									<div class="col-12 col-lg-6 align-self-center">
										<input class="form-control" id="connectedSsid" type="text" disabled>
									</div>

									<div class="col-7 col-lg-4 align-self-center">
										<input class="form-control" id="connectedRssi" type="text" disabled>
									</div>
									<div class="col-5 col-lg-2 align-self-center">
										<button type="button" class="btn btn-danger btn-sm float-right" id="disconnectWifiBt">Disconnect</button>
									</div>
								</div>
								<div class="row" style="margin-top:10px">
									<div class="col-12 col-lg-6 align-self-center">
										<select class="form-control" id="ssid" name="ssid" placeholder="SSID"></select>
									</div>
									<div class="col-12 col-lg-6 align-self-center">
										<input type="text" class="form-control" id="password" placeholder="Password">
									</div>
								</div>
								<div class="row" style="margin-top:10px">
									<div class="col">
										<button type="button" class="btn btn-success btn-sm float-left" id="scanWifiBt">Scan</button>
									</div>
									<div class="col">
										<button type="button" class="btn btn-success btn-sm float-right" id="saveWifiBt">Save</button>
									</div>
								</div>
							</form>
						</div>
					</div>
				</div>
			</div>
		</div>
		<script type="text/javascript" src="/js/jquery-3.3.1.slim.min.js"></script>
		<script type="text/javascript" src="/js/bootstrap.min.js"></script>
		<script type="text/javascript" src="/js/socket.js"></script>
	</body>
</html>
```

Very often it happens that file seems to be loaded, but the page never get generated. I have found that from time to time, if i go in the dev tool and Network in chrome, i see that a random file never the same don'T seems to get fully loaded or the webbrowser get the wrong file size. Here is an image of what i mean.
[image]

Does someone has already seen that?

zekageri, 2021-01-12:
I don't looked into it but if you are using esp32 i recommend to switch to LITTLEFS instead of SPIFFS.
[Here is a link from LITTLEFS by lorol in Github](https://github.com/lorol/LITTLEFS)

Speedy2k, 2021-01-12:
I have just tried it and it does the same problem!! I really don't understand what is going on!

gitpeut, 2021-01-29:
Maybe you run into to maximum number of files the ASync Webserver can serve simultaneously.  In the HTML I count 5 external files, including the html itself that is 6 files that are downloading almost at the same time.
Try to include the javascript and css files in the HTML file and see what happens.
Sometimes it's also possible to load scripts a little later.
Personally I refrain from using frameworks on ESP's to avoid these kinds of issues.

closed 2021-06-16 by stale[bot]