me-no-dev/ESPAsyncWebServer #923: Chunked response hangs at one point, tcp_sent events not handled and triggers WDT

raomin, 2021-01-20:
I store sensor data in a packed binary format on LittleFS. It can be 100k data points.
I have put a `ChunkedResponse` in place retrieve this data in CSV format (the Content-Length is unknown).

Everything starts ok, chunks of data are properly transmitted but when after some time (around 400Kb sent), the traffic stops, something hangs and after `CONFIG_ESP_TASK_WDT_TIMEOUT_S` Task watchdog is triggered.

```log
...
Serving one chunk. Index 411529, maxlen 2861
+S: 0x3ffd1dc8
+S: 0x3ffd1dc8
+P: 0x3ffd1dc8
-S: 0x3ffd1dc8
...
Serving one chunk. Index 428094, maxlen 2890
+S: 0x3ffd1dc8
-S: 0x3ffd1dc8
a:1436:0
Serving one chunk. Index 430787, maxlen 3035
+S: 0x3ffd1dc8
-S: 0x3ffd1dc8
a:1438:0
Serving one chunk. Index 433604, maxlen 2911
+S: 0x3ffd1dc8
-P: 0x3ffd1dc8
p
   [many more like these]
Serving one chunk. Index 477824, maxlen 2884
+S: 0x3ffd1dc8
-S: 0x3ffd1dc8
a:2665:0
Serving one chunk. Index 480487, maxlen 3065
+S: 0x3ffd1dc8
+P: 0x3ffd1dc8
 <--- HERE HANGING FOR TASK_WDT_TIMEOUT seconds

E (6130293) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6130293) task_wdt:  - async_tcp (CPU 0/1)
E (6130293) task_wdt: Tasks currently running:
E (6130293) task_wdt: CPU 0: IDLE0
E (6130293) task_wdt: CPU 1: loopTask
E (6130293) task_wdt: Aborting.
```
If I disable WDT with `CONFIG_ASYNC_TCP_USE_WDT`, the process hangs indefinitely. So I don't think it's a starving / non feeding watchdog issue. More that AsyncTCP and/or ESPAsyncWebServer waits (without yielding) indefinitely for something not happening.

From what I could see in the debugging element that whenever there is a chunk with `+S` `_send_async_event on _tcp_sent`, it looks like _it_ is waiting for `-S` `_handle_async_event LWIP_TCP_SENT` which does not happen.

I expect that the number of `+S` ie `_send_async_event on _tcp_sent` should equal to the number of `-S` ie `_handle_async_event LWIP_TCP_SENT`. 

When the process hangs and WDT is triggered I count **176** `+S` and only **149** `-S`. Which would mean that there are 27 _tcp_sent async event sent but not handled.

Having a look at the number of pending _tcp_sent events over the log line number:
[image]

I'm at my limit of understanding for now this library and this issue.
Could someone help me investigate further and get the chunked response working in this case of intense traffic?

raomin, 2021-01-20:
Increasing the async_queue length delays the issue.
```c 
_async_queue = xQueueCreate(250, sizeof(lwip_event_packet_t *));```

Now this is obviously not a solution but it shows that the WDT is triggered by the async_queue being full of not handled events.

themindfactory, 2021-01-20:
in your debug can you print out the heap and see what that is doing..  I
had similar issues in the past and it was heap....

I made my own chunked code and it worked fine....

RichardS

raomin, 2021-01-20:
Heap remains stable. It crashes with 160Kb free heap.

I moved to LittleFS (I thought I did already but I was still on SPIFFS). 
Computation time in chunk handling is faster on LittleFS, all events seems to be processed correctly and the request completes at more than 3MB.

I believe that there is still something to have a look at and I'd love to understand better what doesn't work when there is less time available for event handling.

themindfactory, 2021-01-20:
Issue with spiffs is sometimes exection times for it skyrocket, I'd blame
it more than async.. Add a small delay to your file accesses to slow it to
spiffs levels and see...

raomin, 2021-01-20:
Yes, you're right, adding a delay(10) would also lead to WDT. Even with a very large queue of 250.

I'll might be safe with LittleFS but what if one really needs to do some process during chunk...?

I would expect that the tasks in queue have higher priority than the normal processing so that TCP events can be properly handled and not stack up despite the processing time taken by one core on the chunk... or do they share the same priority? 🤷‍♂️

themindfactory, 2021-01-20:
When your asked for a chunk and it says it can take 2800  ytes how much do
you give?

closed 2021-06-02 by stale[bot]