me-no-dev/ESPAsyncWebServer #934: Updated documentation for Basic/Advanced Response for ArduinoJson 6.x

LenirSantiago, 2021-02-03:
Below is the updated documentation for ArduinoJson Basic/Advanced Response for ArduinoJson 6.x

### ArduinoJson Basic Response (v6.x)
```
#include "AsyncJson.h"
#include "ArduinoJson.h"

AsyncResponseStream *response = request->beginResponseStream("application/json");
const int JSON_MAX_SIZE = 1024
DynamicJsonDocument doc(JSON_MAX_SIZE );
JsonObject root = doc.to<JsonObject>();
root["heap"] = ESP.getFreeHeap();
root["ssid"] = WiFi.SSID();
serializeJson(root, *response);
request->send(response);
```

### ArduinoJson Advanced Response (v6.x)
```
#include "AsyncJson.h"
#include "ArduinoJson.h"

AsyncJsonResponse * response = new AsyncJsonResponse();
response->addHeader("Server","ESP Async Web Server");
JsonVariant& jsonObj = response->getRoot();

jsonObj["heap"] = ESP.getFreeHeap();
jsonObj["ssid"] = WiFi.SSID();

response->setLength();
request->send(response);
```

AlexLarentis01, 2021-03-02:
Please Put this in the readme

closed 2021-06-16 by stale[bot]

Pezhvak, 2021-10-05:
such a shame this has not been merged

zekageri, 2021-10-06:
Literally nothing is merged

StefanRvO, 2023-07-28:
Bump, it is annoying that the example doesn't work and you have to search for quite some time to figure out how to do this.