me-no-dev/ESPAsyncWebServer #936: Unable to get simple web server working over Ethernet connection

har-in-air, 2021-02-06:
Arduino-ESP32 v 1.0.4, ESP32 
Can ESPAsyncWebServer  work over Ethernet? I have the EthernetWebServer library example running on my ESP32 board connected via SPI to an ENC28J60 Ethernet module. I am able to get a simple "hello world" web server working over the ethernet connection.

When I try to use AsyncWebServer, it crashes and reboots when server.begin() is called.  Prior to that the Ethernet connection is established, am able to get a local IP address allocated via DHCP.
```
#include "defines.h"
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char *PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest *request)
{
    request->send(404, "text/plain", "Not found");
}

void setup(void)
{
    // Open serial communications and wait for port to open:
    Serial.begin(115200);
    while (!Serial)
        ;

    Serial.print("\nStarting AsyncEnetServer on " + String(BOARD_TYPE));
    Serial.println(" with " + String(SHIELD_TYPE));

    // start the ethernet connection and the server:
    // Use DHCP dynamic IP and random mac
    // Use Static IP
    pinMode(4, OUTPUT);
    digitalWrite(4, LOW);
    delay(25);
    digitalWrite(4, HIGH); //enable the ethernet chip after resetting
    //Ethernet.begin(mac, ip);
    Ethernet.begin(mac);
    if (Ethernet.hardwareStatus() == EthernetNoHardware)
    {
        Serial.println(F("Ethernet hardware not found"));
    }

    Serial.println(F("SPI pinout to use"));
    Serial.print(F("MOSI:"));
    Serial.println(MOSI);
    Serial.print(F("MISO:"));
    Serial.println(MISO);
    Serial.print(F("SCK:"));
    Serial.println(SCK);
    Serial.print(F("SS:"));
    Serial.println(SS);

    Serial.print(F("Connected! IP address: "));
    Serial.println(Ethernet.localIP());
    delay(1000);

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        request->send(200, "text/plain", "Hello, world");
    });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [](AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE))
        {
            message = request->getParam(PARAM_MESSAGE)->value();
        }
        else
        {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
    });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE, true))
        {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        }
        else
        {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
    });

    server.onNotFound(notFound);
    server.begin();
}

void loop(void)
{
}
```
```
#ifndef defines_h
#define defines_h

#define DEBUG_ETHERNET_WEBSERVER_PORT Serial
// Debug Level from 0 to 4
#define _ETHERNET_WEBSERVER_LOGLEVEL_ 3
// For ESP32
#warning Use ESP32 architecture
#define ETHERNET_USE_ESP32
#define BOARD_TYPE "Synectix"

#include <SPI.h>

// Use true  for ENC28J60 and UIPEthernet library (https://github.com/UIPEthernet/UIPEthernet)
#define USE_UIP_ETHERNET true

#if USE_UIP_ETHERNET
#include "UIPEthernet.h"
#warning Using UIPEthernet library
#define SHIELD_TYPE "ENC28J60 using UIPEthernet Library"
#endif 

//#include <EthernetWebServer.h>

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x01};
// if static IP address used (change as per local network)
//IPAddress ip(192, 168, 2, 222);

#endif //defines_h
```

Serial debug output ...

```
Starting AsyncEnetServer on Synectix with ENC28J60 using UIPEthernet Library
SPI pinout to use
MOSI:23
MISO:19
SCK:18
SS:5
Connected! IP address: 10.42.0.251
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088071 on core 1
```

har-in-air, 2021-02-06:
I decoded the backtrace using EspExceptionDecoder
```
Backtrace: 0x4008b744:0x3ffb1d90 0x4008b975:0x3ffb1db0 0x40088071:0x3ffb1dd0 0x400fda36:0x3ffb1e10 0x400fdb12:0x3ffb1e30 0x400f24e0:0x3ffb1e50 0x400f2549:0x3ffb1e70 0x400f3aa0:0x3ffb1e90 0x400f3b8c:0x3ffb1ec0 0x400d54c9:0x3ffb1ee0 0x400da506:0x3ffb1f30 0x400d1963:0x3ffb1f50 0x400dc2b3:0x3ffb1fb0 0x40088385:0x3ffb1fd0
```
```
Decoding stack results
0x4008b744: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008b975: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40088071: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
0x400fda36: mld6_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/mld6.c line 583
0x400fdb12: mld6_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/mld6.c line 244
0x400f24e0: _svfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1586
0x400f2549: _svfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1607
0x400f3aa0: tcp_pcb_remove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1964
0x400f3b8c: tcp_close_shutdown at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 275
0x400d54c9: AsyncClient::add(char const*, unsigned int, unsigned char) at /home/hari/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp line 749
0x400da506: AsyncCallbackWebHandler::canHandle(AsyncWebServerRequest*) at /home/hari/Arduino/libraries/ESPAsyncWebServer/src/WebHandlerImpl.h line 121
0x400d1963: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  tmp/arduino_build_471450/AsyncEnetServer.ino.elf, CU 0x0, DIE 0xc6a5>) at /home/hari/work/synectix/AsyncEnetServer/AsyncEnetServer.ino line 74
0x400dc2b3: String::concat(char const*) at /home/hari/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/WString.cpp line 346
0x40088385: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

woodlist, 2021-03-02:
Almost the same happens in my project. I have used GSM module SIM800L. The test sketch runs smooth until server.begin().
09:46:47.143 -> Waiting for network... success
09:46:47.143 -> Network connected
09:46:47.143 -> Local IP:10.105.64.189
09:46:47.211 -> Signal quality:31
09:46:47.211 -> assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 416, function: tcpip_api_call
09:46:47.211 -> abort() was called at PC 0x400e156b on core 1
09:46:47.211 -> 
09:46:47.211 -> ELF file SHA256: 0000000000000000
09:46:47.211 -> 
09:46:47.211 -> Backtrace: 0x40085908:0x3ffb1e00 0x40085b85:0x3ffb1e20 0x400e156b:0x3ffb1e40 0x400f1b5f:0x3ffb1e70 0x400d7c48:0x3ffb1ea0 0x400dc772:0x3ffb1ef0 0x400d3d4f:0x3ffb1f10 0x400def86:0x3ffb1fb0 0x40086b86:0x3ffb1fd0
09:46:47.211 -> 
09:46:47.211 -> Rebooting...

einored, 2021-03-09:
same issue here, using a prodino esp32. esp reboots on server.begin();

bkaraceylan, 2021-04-19:
Any news on this?

MariusVB, 2021-04-28:
Just encountered the same error here. Downgrading to earlier versions did not seem to fix it.

MariusVB, 2021-04-28:
I'm not using WiFi instead of Ethernet, but encountered the same problem. Just leaving my fix here for anyone who might encounter the same error. 
I fixed it by calling `WiFi.begin()`  or `WiFi.softAP()` before calling `server.begin()` . 

Perhaps your problems can be fixed by creating a unused/hidden softAP before calling `server.begin()`?

closed 2021-07-13 by stale[bot]

videojedi, 2022-01-13:
Having same issue. Has there been any update?