me-no-dev/ESPAsyncWebServer #937: AP connection issues from different devices

dtheodor79, 2021-02-08:
Hello, I have tried the simple web server example, but modified to act as an AP (the code is below for reference). The server runs on an ESP32 board.

After few days of tries, I concluded to the following observations:

The webserver interacts successfully with the following devices:

- Android Xiaomi Redmi 4 + Chrome
- Toshiba laptop + Win10 +  Firefox

Connection issues with the following devices:

- Android Xiaomi Redmi 8 + Chrome -> endless loop of connect / disconnect state (I can see it at the smartphone screen, where the connection stays for 3-4 secs, then it gets dropped).
- Moto G5 + Chrome / Firefox -> connection successful, but page never gets loaded on the smartphone (ERR_CONNECTION_TIMED_OUT). 

Does anyone else have observed the same behavior? Are there any hints on why this is happening? After checking for potential solutions, I ran into [this post](https://rntlab.com/question/solved-esp32-web-server-drops-connection-crashes/) but instead uses the WiFiServer.

Thanks,
dtheodor79

```
//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>

AsyncWebServer server(8080);

const char* ssid = "ESP";
const char* password = "123";

const char* PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/html", "not found");
}

void setup() {

    Serial.begin(57600);
    WiFi.mode(WIFI_AP);
    WiFi.softAP(ssid,NULL,1,0,1);    
    /*WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }*/    

    Serial.print("IP Address: ");Serial.println(WiFi.softAPIP());

    
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [] (AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE)) {
            message = request->getParam(PARAM_MESSAGE)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
    });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request){
        String message;
        if (request->hasParam(PARAM_MESSAGE, true)) {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
    });

    server.onNotFound(notFound);

    server.begin();
}

void loop() {
}
```

wladwnt, 2021-03-13:
What arduino-esp32 version do you use? If I use 1.0.4 (idf 3.2 based) is everything fine, but when I use 1.0.5 (idf 3.3 based) I have problems connecting my Windows 10 PC (Firefox as a browser) to ESP32 AP. The connection drops many times, web pages are laoded very very slowly. With Samsung Galaxy A8 Android Smartphone and Firefox however, is everything OK.

wladwnt, 2021-03-14:
I identified my problem, it was not WebServer but ArduinoOTA. if I removeArduinoOTA.handle() from main loop, everything fine with esp32-arduino 1.0.5.

dtheodor79, 2021-03-16:
I was using 1.0.4 via the Arduino IDE, and now updated to 1.0.5. I didn't use any OTA functionality, so I will try now again and hopefully these instabilities will be gone..

closed 2021-06-16 by stale[bot]