me-no-dev/ESPAsyncWebServer #941: ESPAsync + Captive + LittleFS + processor function = not working with Captive Portal

duckylock, 2021-02-15:
Hi team, 
I use the captive portale exemple and I tried to replace de AsyncResponseStream technic by the external file technic
but it seams to not working for the captive portal.. :'(
(ESP8622 2.7.3 + ESP Async WebServer 1.2.3)

`class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {}
  virtual ~CaptiveRequestHandler() {}

  bool canHandle(AsyncWebServerRequest *request){
    //request->addInterestingHeader("ANY");
    return true;
  }
  
  void handleRequest(AsyncWebServerRequest *request) {
    request->send(LittleFS, "/index.html", String(), false, processor);
  }
};`

and here is my setup() function
`void setup() {
  Serial.begin(115200);
  LittleFS.begin()
  
  dnsServer.setErrorReplyCode(DNSReplyCode::NoError);
  dnsServer.start(DNS_PORT, "*", softAP_ip);

  WiFi.softAPConfig(softAP_ip, softAP_ip, netMsk);
  WiFi.softAP(softAP_ssid, softAP_password);

  //WEBSOCKETSERVER
  wse.onEvent(onWsEvent);
  serverAPoint.addHandler(&wse);
  serverAPoint.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);//only when requested from AP
  serverAPoint.begin(); // Web server start
}
`
it only works with "text/plain" or the AsyncResponseStream object...
Can you comfirm me that is not a bug and if you can explain me why it doesn t work with externel file and the processor function.
thank you

closed 2021-06-16 by stale[bot]

ansonschall, 2022-01-26:
I ran into this issue too. My best guess is that a chunked response does not result in automatic display of a captive portal window.

The documentation for ESPAsyncWebServer states that template processing always results in a chunked response. So, the handler for the request made immediately after joining the ESP WiFi network must avoid template processing. It's worth noting that an HTTP redirect to a route that uses template processing will have the same problem.

My workaround is to serve a simple page without using template processing, which includes a meta tag that results in a redirect to a route that does use template processing: `<meta http-equiv="refresh" content="0;URL='/'">`

looxonline, 2023-02-09:
Can confirm that a redirect works wonders. Just make that the captive landing page and disable processing for the captive handler and it all just works.