me-no-dev/ESPAsyncWebServer #948: Very slow callbacks on chunked responses

HoverClub, 2021-02-24:
My test code supplies log data (via the chunked response callback) every 1sec or so using a timer.  When fresh data isn't available it responds with RESPONSE_TRY_AGAIN to the callback.  When it does so, the callback isn't then called by AsyncWebServer again for exactly 0.5seconds (and, if RESPONSE_TRY_AGAIN, every 0.5secs after that).  When data is provided (any amount) the next callback occurs within a few mSecs.  The consistency of the delay would would indicate that there is some timer somewhere in AsyncWebServer when it gives up on the next chunked response for a fixed period before retrying?  If there is a timeout, does anyone know if it is exposed (I've poked through the lib but can't understand/see anything!)?

My "real" code logs data from server to a file on the client using the normal browser file download mechanism but with chunked response started/stopped by browser (I send small amounts of data every 20mSec or so and see odd 0.5sec delays when data wasn't called for).  I could buffer data on the server (it's read from sensors) but the unpredictability of the transfer delay is a concern (how much would I need to buffer?).  The "slow" file download works quite well - the browser will sit there for hours downloading a file provided there is some data being transferred at regular intervals.

Typical sequence in the callback handler with my 1sec test code (including mSec call time) is:

 call 61977  -> data supplied

 call 61979  -> RESPONSE_TRY_AGAIN
 call 62477  -> RESPONSE_TRY_AGAIN
 call 62977  -> RESPONSE_TRY_AGAIN
 call 63477  -> data supplied

 call 63479  -> RESPONSE_TRY_AGAIN
 call 63977  -> RESPONSE_TRY_AGAIN
 call 64477  -> RESPONSE_TRY_AGAIN
 call 64977  -> data supplied

 call 64979  -> RESPONSE_TRY_AGAIN
 call 65477  -> RESPONSE_TRY_AGAIN
 call 65977  -> RESPONSE_TRY_AGAIN
 call 66477  -> data supplied
...

closed 2021-06-16 by stale[bot]

Mark42XLII, 2021-07-13:
I have the same issue over here. Tried raising an user exception right before returning `RESPONSE_TRY_AGAIN` but the exception decoder shows a bunch of lwip functions and literaly none related to this library.

stas-sl, 2024-10-15:
It appears that the chunked response callback is triggered by two TCP events:

1. ACK: This event occurs when the previously sent data has been successfully delivered. It happens when you return non-empty data from your previous callback invocation.
2. POLL: This happens periodically, at an interval specific to your TCP connection, which is initialized in the `AsyncClient` constructor. By default, this polling event occurs twice a second.

Here is an excerpt from the lwIP `tcp_poll` [documentation](https://www.nongnu.org/lwip/2_1_x/group__tcp__raw.html#gafba47015098ed7ce523dcf7bdf70f7e5):

> Specifies the polling interval and the callback function that should be called to poll the application. The interval is specified in terms of the TCP coarse-grained timer, which typically fires twice per second. For example, an interval of 10 would mean that the application is polled every 5 seconds.

> When a connection remains idle (i.e., no data is being sent or received), lwIP will repeatedly poll the application using the specified callback function. This feature can act as a watchdog timer to terminate idle connections or serve as a mechanism to wait for memory to become available. For instance, if a call to tcp_write failed due to insufficient memory, the polling callback could attempt another tcp_write() after a period of inactivity.

I can think of several workaround hacks:
1. Call `tcp_poll` on an existing client connection passing new poll interval
2. Manually call `AsyncChunkedResponse::_ack` (which is public despite name starting with underscore) from somewhere as frequent as you need

I haven't tested these methods myself, so I can't guarantee they'll work or predict the effects of more frequent poll events. However, they are simply some suggestions to consider.