me-no-dev/ESPAsyncWebServer #951: Use-after-free in AsyncWebServerRequest::_removeNotInterestingHeaders

Depau, 2021-03-07:
Hi,
While running my code with Valgrind in an effort to troubleshoot memory-corruption issues I noticed there's a use-after-free error in this library's code.

Here the function iterates over a linked list of headers, and removes not-interesting ones from the list.
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L183-L186

Here, the linked list's remove method correctly updates the next pointer and frees the removed item:

https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/StringArray.h#L120-L132

However, if an item is deleted during iteration, the iterator still holds a reference to its memory, and it still uses it to retrieve the next pointer:

https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/StringArray.h#L53

While this works in most cases and particularly on the ESP MCUs, on my testing setup I get a nice SIGSEGV.

A potential fix could be retrieving the `next` pointer ahead of time and storing it in the iterator. Then in `operator++` replace it with the "next-next" pointer and return the stored `next` pointer. This should make deleting the current item safe.

---
<small>
(If you're wondering how the hell I'm running it with valgrind, I stubbed all the Arduino calls and patched ESPAsyncTCP to use the POSIX socket API, and I'm running my application as a regular Linux program)
</small>

zekageri, 2021-05-25:
Do you have a fork to try with the fixes?

Depau, 2021-05-25:
This branch has all my pull requests:
https://github.com/Depau/ESPAsyncWebServer/tree/wi-se-patches

zekageri, 2021-05-26:
Thanks. I found one before but I think it wasnt that one. I will try it. Iam experiencing random crashes on my esp and it seems it is because this linked list problem. I cant debug because its remote and I cant put serial to it

Depau, 2021-05-31:
If you'd like to run your ESP application on a Linux computer, I wrote a bunch of stubs to do so: https://github.com/Depau/wi-se-sw/tree/main/fakeesp This is what allowed me to run this library with Valgrind/GDB and find the issues.

YMMV though, and I don't have time to provide any supports on them. Feel free to do whatever you want with the code, if you have any improvements lmk.

zekageri, 2021-06-01:
Thank you for the help. Your version of the library seems pretty stable. It runs almost 4 days straight now. 
I got a crash like in 20 hours before. But i really get stable consistent file servings and client connection/disconnetions. I will look at your linux thingy too.
I really appreciate these and hope Me_no_Dev will see your fixes

Depau, 2021-08-01:
Unstable, still waiting for this to be reviewed

brandonros, 2021-08-16:
@Depau 

```
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(AsyncWebSocket.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x10): undefined reference to `SHA1Init'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(AsyncWebSocket.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x14): undefined reference to `SHA1Update'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(AsyncWebSocket.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x18): undefined reference to `SHA1Final'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(AsyncWebSocket.o): in function `String::operator+=(char const*)':
/Users/brandonros/Desktop/esp32-bench-flasher/components/arduino-esp32/cores/esp32/WString.h:128: undefined reference to `SHA1Init'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(AsyncWebSocket.o): in function `String::len() const':
/Users/brandonros/Desktop/esp32-bench-flasher/components/arduino-esp32/cores/esp32/WString.h:305: undefined reference to `SHA1Update'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(AsyncWebSocket.o): in function `AsyncWebSocketResponse::AsyncWebSocketResponse(String const&, AsyncWebSocket*)':
/Users/brandonros/Desktop/esp32-bench-flasher/components/ESPAsyncWebServer/src/AsyncWebSocket.cpp:1341: undefined reference to `SHA1Final'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(WebAuthentication.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
/Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/brandonros/Desktop/esp32-bench-flasher/build/ESPAsyncWebServer/libESPAsyncWebServer.a(WebAuthentication.o): in function `getMD5(unsigned char*, unsigned short, char*)':
/Users/brandonros/Desktop/esp32-bench-flasher/components/ESPAsyncWebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'

```

when trying to use your branch

brandonros, 2021-08-16:
@Depau can you please pull latest `me-no-dev` master into your ``wi-se-patches` branch and push? It does not build otherwise.

Depau, 2022-04-05:
Unstale

szbeni, 2022-05-11:
Is this bug fix going to be merged?

softhack007, 2022-09-18:
Is this bugfix already integrated?

Depau, 2022-09-18:
> Is this bugfix already integrated?

No

softhack007, 2022-09-18:
@me-no-dev Is it planned to integrate this fix? 
We are currently trying to port our software (WLED) to the new MCU's, and this one seems to cause troubles on -C3 and (maybe) -S2.

A40in, 2024-10-17:
replace in  WebRequest.cpp
```cpp
void AsyncWebServerRequest::_removeNotInterestingHeaders(){
  if (_interestingHeaders.containsIgnoreCase("ANY")) return; // nothing to do
  StringArray *a = &_interestingHeaders;
  auto cmpr = LinkedList<AsyncWebHeader*>::Predicate{[&a](AsyncWebHeader* h) {
    return a->containsIgnoreCase(h->name().c_str());
  }};
  while (_headers.remove_first(cmpr)){}
}
```
or
```cpp
void AsyncWebServerRequest::_removeNotInterestingHeaders(){
  if (_interestingHeaders.containsIgnoreCase("ANY")) return; // nothing to do
  StringArray *a = &_interestingHeaders;
  auto cmpr = LinkedList<AsyncWebHeader*>::Predicate{[&a](AsyncWebHeader* h) {
    return a->containsIgnoreCase(h->name().c_str());
  }};
  _headers.remove_all(cmpr);
}
```
and add metod  to class LinkedList  in StringArray.h
```cpp
    bool remove_all(Predicate predicate){
      auto it = _root;
      auto pit = _root;
      bool flag = false;
      while (it) {
        if(predicate(it->value())){
          if(it == _root){
            _root = _root->next;
          } else {
            pit->next = it->next;
          }
          if (_onRemove) {
            _onRemove(it->value());
          }
          delete it;
          it = pit->next;
          flag = true;
        } else {
          pit = it;
          it = it->next;
        }
      }
      return flag;
    }

```