me-no-dev/ESPAsyncWebServer #957: How to connect to WebSocket as client in ESP32?

augustoeliezer, 2021-03-16:
In [AsyncWebSocketClient ](https://github.com/me-no-dev/ESPAsyncWebServer#methods-for-sending-data-to-a-socket-client) have a client connected to AsyncWebSocket. But how to connect as client in ESP32?
**Server.ino**
```c#
void setup() {
//...
  AsyncWebSocket ws("/ws");
  ws.onEvent(onEvent);
  server.addHandler(&ws);
  //...
}
```

### TODO
**Client.ino**
```c#
void setup() {
  //...
  AsyncWebSocketClientRequest wsc("ws://192.168.4.1:80/ws");
  
  if (wsc.connect()) {
  
    wsc.onData([](String data) {
  
    });
  }
  //...
}
```

augustoeliezer, 2021-06-03:
Any suggestion?

jendakol, 2021-06-03:
Hi @augustoeliezer, it doesn't work for me either. I had to use another WS client, see [my code](https://github.com/jendakol/lasery/blob/master/sensor/src/main.cpp#L47).

Squanchy42, 2021-07-14:
I know this is a rather old issue, but for future reference maybe try using ArduinoWebsockets on the client side. With a static ip on the server side you can access your websocket by "ws://your_static_ip:port/ws". If you stick with the example code snippets your port number is likely 80.