me-no-dev/ESPAsyncWebServer #963: Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

tonydee2377, 2021-03-24:
Hi,
I wanted to bring this frequent AsyncEventSourceClient error to attention:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d9aa4  PS      : 0x00060630  A0      : 0x800d9b42  A1      : 0x3ffb1ef0  
A2      : 0x3ffd3f14  A3      : 0x3ffd6b74  A4      : 0x00000026  A5      : 0x3ffd4026  
A6      : 0x00000a0d  A7      : 0x3a617461  A8      : 0x800d9ab0  A9      : 0x3ffb1ed0  
A10     : 0x00000000  A11     : 0x0000004c  A12     : 0x00000026  A13     : 0x00000026  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d9aa4:0x3ffb1ef0 0x400d9b3f:0x3ffb1f10 0x400d9b5f:0x3ffb1f30 0x400d9bb5:0x3ffb1f50 0x400d505f:0x3ffb1f80 0x400e103d:0x3ffb1fb0 0x40089ae6:0x3ffb1fd0

Rebooting...

PC: 0x400d9aa4: AsyncEventSourceClient::_runQueue() at C:\ProgrammiP\arduino-1.8.13\portable\sketchbook\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 244
EXCVADDR: 0x00000008

Decoding stack results
0x400d9aa4: AsyncEventSourceClient::_runQueue() at C:\ProgrammiP\arduino-1.8.13\portable\sketchbook\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 244
0x400d9b3f: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at C:\ProgrammiP\arduino-1.8.13\portable\sketchbook\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 194
0x400d9b5f: AsyncEventSourceClient::write(char const*, unsigned int) at C:\ProgrammiP\arduino-1.8.13\portable\sketchbook\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 229
0x400d9bb5: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at C:\ProgrammiP\arduino-1.8.13\portable\sketchbook\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 320
0x400d505f: loop() at C:\Users\ANTONIO\Desktop\Antonio Ciaravola\new project\MOS32/MOS32.ino line 677
0x400e103d: loopTask(void*) at C:\ProgrammiP\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.5\cores\esp32\main.cpp line 37
0x40089ae6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

I solved by lowering the refresh in this loop, that if set to 1s the core 1 crashes, but setting it to 2000 the problem vanishes.
For more frequent http updates we have to wait for a library update. I hope it will be of help to someone. 

Bye

tonydee2377, 2021-03-24:
ah, here's the crash routine:

static const unsigned long EVENT_INTERVAL_MS = 2000;     //Refresh value 2s   <------(1000 causes the crash )

 if ((millis() - lastEventTime) > EVENT_INTERVAL_MS) {      
      events.send("ping", NULL, millis());
      events.send(String(Scatlav).c_str(), "lavorate", millis());
      events.send(String(Scatalm).c_str(), "al minuto", millis());
    //  events.send(String("123456").c_str(),"spostate",millis());
      events.send(String(Velmin).c_str(), "mminuto", millis());
      events.send(String(invState).c_str(), "inv_state", millis());
      events.send(String(stpState).c_str(), "stp_state", millis());
      events.send(String(selectprog).c_str(), "program", millis());
    lastEventTime = millis();
  }

ismal, 2021-04-01:
why not to send everything as one string (for example, coma separated) with one events.send ? For me if solved the problem with crushes completely.

zekageri, 2021-04-10:
It happens with websocket too, not just events. :/

closed 2021-06-24 by stale[bot]