me-no-dev/ESPAsyncWebServer #965: Guru Meditation Error when more than one client is connected

DShabroff, 2021-04-10:
I get Guru Meditation Error when more than one client is connected.

EXCVADDR: 0x00000010

Decoding stack results
0x400df80a: AsyncEventSourceClient::_runQueue() at D:\ESP32_Work\Soft\libraries\ESPAsyncWebServer-master\src\AsyncEventSource.cpp line 242
0x400df8a7: AsyncEventSourceClient::write(char const*, unsigned int) at D:\ESP32_Work\Soft\libraries\ESPAsyncWebServer-master\src\AsyncEventSource.cpp line 229
0x400df8c7: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at D:\ESP32_Work\Soft\libraries\ESPAsyncWebServer-master\src\AsyncEventSource.cpp line 317
0x400df91d: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at D:\ESP32_Work\Soft\libraries\ESPAsyncWebServer-master\src\AsyncEventSource.cpp line 323
0x400d296d: AsyncServerOperate() at D:\ESP32_Work\Soft\sketch_apr09a/sketch_apr09a.ino line 604
0x400d3349: loop() at D:\ESP32_Work\Soft\sketch_apr09a/sketch_apr09a.ino line 701
0x400e6c08: app_main() at C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 29
0x4009073a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Please help me. Thanks.

xlyric, 2021-04-12:
same

PC: 0x4000c29b
EXCVADDR: 0xbaad5678

Decoding stack results
0x400dab22: AsyncAbstractResponse::_fillBufferAndProcessTemplates(unsigned char*, unsigned int) at .pio\libdeps\esp32dev\ESP Async WebServer\src\WebResponses.cpp line 455
0x400dad25: AsyncAbstractResponse::_ack(AsyncWebServerRequest*, unsigned int, unsigned int) at .pio\libdeps\esp32dev\ESP Async WebServer\src\WebResponses.cpp line 313
0x4018a6be: AsyncWebServerRequest::_onAck(unsigned int, unsigned int) at .pio\libdeps\esp32dev\ESP Async WebServer\src\WebRequest.cpp line 201
0x400d6f11: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ,  ,  ) at .pio\libdeps\esp32dev\ESP Async WebServer\src\WebRequest.cpp line 73
0x4017590b: AsyncClient::_sent(tcp_pcb*, unsigned short) at c:\users\c_lyr\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x40175a74: AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at .pio\libdeps\esp32dev\AsyncTCP\src\AsyncTCP.cpp line 1203
0x401760c5: _async_service_task(void*) at .pio\libdeps\esp32dev\AsyncTCP\src\AsyncTCP.cpp line 165
0x4008a30e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

zekageri, 2021-04-13:
I belive this is the same problem because i applied the patched up versions mentioned in this issue #932 
and it seems to me that it works so far. But additional test are required

DShabroff, 2021-04-14:
It took some time for testing. Reboots have become less frequent. I downloaded 2 server pages on one computer, and 3 on another. The reboot now occurs once in 4-6 hours and does not depend on the number of open copies. I haven't tested it with one open copy, but I will test it and writeabout.  Below decoded error code

Decoding stack results
0x4008f448: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008f6c5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x401d1a6b: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401d1ab2: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x401d13d9: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x401d14b4: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400df34d: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at E:\Prog\libraries\ESPAsyncWebServer-dev\src/StringArray.h line 67
0x400df39b: AsyncEventSourceClient::_write(char const*, unsigned int) at E:\Prog\libraries\ESPAsyncWebServer-dev\src\AsyncEventSource.cpp line 240
0x400dfa02: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at E:\Prog\libraries\ESPAsyncWebServer-dev\src\AsyncEventSource.cpp line 343
0x400d2a51: AsyncServerOperate() at E:\Prog\ADXL3xx/ADXL3xx.ino line 630
0x400d3462: loop() at E:\Prog\ADXL3xx/ADXL3xx.ino line 735
0x400e6ed0: loopTask(void*) at C:\Users\Dima\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x4009073a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

zekageri, 2021-04-14:
Does your client reconnects on websocket if its lost?

DShabroff, 2021-04-14:
> Does your client reconnects on websocket if its lost?

Yes, sometimes I see the massage: Client reconnected with ID that it got is...

DShabroff, 2021-04-15:
In the case of one open page, no processor resets are observed. As it was before.

xlyric, 2021-04-15:
if it can help :   
personnaly the esp32 hang with the mode "processor" 
request->send(SPIFFS, "/index.html", String(), false, processor); 
and work only with    
request->send(SPIFFS, "/index.html", "text/html");
I processing data after

DShabroff, 2021-04-15:
> request->send(SPIFFS, "/index.html", "text/html");
It's not compilng yet. In my code Processor - is the simple function:

`String processor(const String& var)
{
    return  "Waiting Data";
}
`
If I write function without arguments, I get compile error:

> ADXL3xx:503:60: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
request->send_P(200, "text/html", index_html, processor);
Same compile eror I get with this code:
`request->send_P(200, "text/html", index_html, "Waiting Data");`

This variant:
`    String PageWait = "Waiting Data";
    request->send_P(200, "text/html", index_html, PageWait);`

get compile eror is:

> ADXL3xx:504:59: error: no matching function for call to 'AsyncWebServerRequest::send_P(int, const char [10], const char [2648], String&)'
     request->send_P(200, "text/html", index_html, PageWait);
I think last argument in request->send_P(200, "text/html", index_html, PageWait);` must be the function.

Pablo2048, 2021-04-15:
Yes, the last argument is pointer to the function. Try to use:
```cpp
String processor(const String& var) { return String("Waiting Data"); }
```

DShabroff, 2021-04-15:
> ```c++
> String processor(const String& var) { return String("Waiting Data"); }
> ```
I think it's same as I use:

> String processor(const String& var) { return "Waiting Data"; }

But I will try your variant. Thank you.

Pablo2048, 2021-04-15:
> I think it's same as I use:
> 
> > String processor(const String& var) { return "Waiting Data"; }
> 
It's not - Your function return `const char *`, while required return is String() object...

DShabroff, 2021-04-17:
Reset are rare but still occur. Once per 20 hours of work.  4 pages open.

`Decoding stack results
0x4008f448: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008f6c5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x401d1caf: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401d1cf6: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x401d161d: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x401d16f8: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400df595: AsyncEventSourceClient::_write(char const*, unsigned int) at E:\Mustor\libraries\ESPAsyncWebServer-dev\src\AsyncEventSource.cpp line 240
0x400dfc0e: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at E:\Mustor\libraries\ESPAsyncWebServer-dev\src\AsyncEventSource.cpp line 343
0x400d299d: AsyncServerOperate() at E:\Mustor\ADXL3xx/ADXL3xx.ino line 655
0x400d366e: loop() at E:\Mustor\ADXL3xx/ADXL3xx.ino line 873
0x400e7104: loopTask(void*) at C:\Users\Dima\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x4009073a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

zekageri, 2021-04-21:
If no reset, the server is frozen for me. The esp still working doing its job but i cant reach the web. I experience this with the fixed libs.

closed 2021-07-08 by stale[bot]